FreeCAD/FreeCAD #23888: No write permission error when exporting an object to OneDrive folder on Windows

philip-marlowe, 2025-09-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Exporting any object to a folder under OneDrive fails with 'No write permission for file' error.

From FreeCAD.log
> Err: <Exception> No write permission for file
> Err: <class 'SystemError'>: unknown opcode
> Wrn: unknown opcodeErr: Err: SystemErrorErr: : Err: unknown opcodeErr: 

Saving a FCStd file to the same location works without a problem.
The same operation worked till 2025.09.03 build.

FreeCAD_weekly-2025.09.03-Windows-x86_64-py311: OK
FreeCAD_weekly-2025.09.10-Windows-x86_64-py311: NG
FreeCAD_weekly-2025.09.11-Windows-x86_64-py311: NG
FreeCAD_weekly-2025.09.12-Windows-x86_64-py311: NG

C:\Users\<USERNAME> : OK
C:\Users\<USERNAME>\OneDrive : NG
C:\Users\<USERNAME>\OneDrive\Documents: NG
C:\Users\<USERNAME>\OneDrive\Downloads: NG

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/12 03:41:38
Build type: Release
Branch: main
Hash: 952bff5b63a62b55a7ea1623edfb39e90dee2d4c
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Japanese/Japan (ja_JP)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/99.4147/1.5
Installed mods: 
  * Curves 0.6.64
  * sheetmetal 0.7.24
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-09-15:
I cannot reproduce in the latest weekly. @chennes do you have Win + OneDrive?

philip-marlowe, 2025-09-17:
This problem persists on the latest weekly build.
At least in my test, there is no dependency on export file type.

FreeCAD_weekly-2025.09.17-Windows-x86_64-py311

ImBehinYa, 2025-09-21:
i have the same issue,but every folder doesnt work

chennes, 2025-09-21:
It's very strange -- it's not really limited to OneDrive, but I can't figure out what the pattern is. `std::filesystem` is reporting read-only permissions for OneDrive, but also for the "C:\users\chennes\Saved Games", even though I can create files there. I thought maybe it was the space in the name, but I tried a bunch of different exotic folder names and none of the others gave me any problems.

chennes, 2025-09-21:
My bet is that this check:
```
    return (perms & fs::perms::owner_write) == fs::perms::owner_write;
```
is inadequate. We may need something like:
```
(p & fs::perms::owner_write) != fs::perms::none ||
        (p & fs::perms::group_write) != fs::perms::none ||
        (p & fs::perms::others_write) != fs::perms::none
```
and even that is not really the whole store (my understanding is that it ignores ACLs, for example). Probably if we really want "bullet-proof" checking, we should just try to make a dummy file and handle any exceptions that occur.

ebswift, 2025-09-25:
Like others reported, on Windows 11 I get the issue on the latest weekly 2025.09.24.  While I initially noticed it on a locally sync'd google drive location, I still get the issue if I save outside of that drive location to a local unmanaged folder.

Specifically, I'm seeing this:

- In versions I'm having trouble with, I can export a STEP but I cannot export an STL
- Attempting to export to another folder doesn't help
- I can successfully export the STL in weekly build 2025.09.03
- I get the no write permission error if I attempt to export using weekly build 2025.09.11

So, hopefully narrowed down between the 2025.09.03 windows weekly build and 2025.09.11 windows weekly build.

closed 2025-10-06 by sliptonic