FreeCAD/FreeCAD #23889: Sketcher: Canceling BSpline at the first point leave the point

leoheck, 2025-09-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Esc after the first click will keep the first point. 

https://github.com/user-attachments/assets/bcd059c6-d466-4ea3-a46d-11099e0ac56c

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/12 03:41:38
Build type: Release
Branch: main
Hash: 952bff5b63a62b55a7ea1623edfb39e90dee2d4c
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
Installed mods: 
  * Assembly4.1 0.60.2.1
  * Cables 0.2.1
  * CurvesWB 0.6.64
  * FreeCAD-render 2024.12.15
  * sheetmetal 0.7.52
  * DynamicData 2.75.0
  * Curves 0.6.70
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2025.5.20
  * fasteners 0.5.41
  * Silk 0.2.0
  * osh-autodoc-workbench 0.2.3
  * CurvedShapes 1.0.13
  * freecad-povray-render
  * OpticsWorkbench 1.2.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.7.6
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

matthiasdanner, 2025-09-13:
I can confirm that, but for me it only happens after I completed a bspline and deleted it. The bug persists if I close and reopen the sketch. But when i delete the sketch and create a new one the handling is reset, and the point is deleted on cancel.

https://github.com/user-attachments/assets/b89efd12-3f1e-49bd-b4e4-498609dc4d3a

PaxTux, 2026-03-24:
Still relevant on 1.1.0rc3 (followed steps from @matthiasdanner).
Cannot reproduce on latest 1.2.0dev.

matthiasdanner, 2026-04-13:
It changed a Bit of the State handling in sketches. So it was a Nice side effect 😁