FreeCAD/FreeCAD #23895: Ctrl+A shortcut does not work on the Report view

furgo16, 2025-09-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

A Report view with some content (e.g. log, warning or error messages) is assumed
1. Open the Report view
2. Click on an empty area of the Report view panel to ensure it has focus
3. Press Ctrl+A to select all contents
4. **Expected**: the contents of the Report view are selected
5. **Actual**: the contents of the Tree view are selected

If instead on step 3 the Report view's context menu is open with right click, and then the `Select All` menu option there is chosen, the Report view's contents are selected.

It would seem that the tree view is stealing the Ctrl+A shortcut.

### Full version info

```shell
1.1.0-dev
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-09-13:
Looks like a recent regression, was working in August using:

```
OS: Linux Mint 22.2 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.43003 (Git) AppImage
Build date: 2025/08/22 17:47:02
Build type: Release
Branch: main
Hash: 26a7067e44a27003bc013a96bd37183d20b62dd0
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Window center
Stylesheet/Theme/QtStyle: Behave-dark_Pre1_0.qss/unset/
Logical DPI/Physical DPI/Pixel Ratio: 96/93.1131/1
Installed mods: 
  * FreeCAD-Ribbon
  * Manipulator 1.6.0
  * FreeCAD-themes 2025.1.7
  * Defeaturing 1.2.6
  * sheetmetal 0.7.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.31.8
  * fasteners 0.5.39
  * AddonManager 2025.7.17
```

Syres916, 2025-09-13:
Also your steps are missing the fact that the user must have a document open before clicking on the Report View.

Syres916, 2025-09-13:
OK so I've narrowed it down to between 24th August and 1st September, Bisect running and will report back.

furgo16, 2025-09-13:
I've added a note to indicate an open document is required to reproduce, thanks!

Throwing a wild guess, which matches your time window, there was this PR that touched Ctrl+A: https://github.com/FreeCAD/FreeCAD/pull/23289

Syres916, 2025-09-13:
You are bang on with https://github.com/FreeCAD/FreeCAD/pull/23289

On my 1st Sept snapshot, I ran:

`git revert 3b76d77ed8ab31ffeca69064e2925f17774f087c`

recompiled and sure enough `Ctrl+A` shortcut on the Report View with a document open works again.

furgo16, 2025-09-13:
@tetektoza @PaddleStroke (as you worked on follow-up fixes in https://github.com/FreeCAD/FreeCAD/issues/23863) FYI

closed 2025-09-13 by kadet1090