FreeCAD/FreeCAD #23901: Assembly: Adding the limits for the joints will break the joints

Conscientiia, 2025-09-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Wish you guys have a good day.

Today I experienced another issue relating to the Assembly WB is that when you add the "Min" and "Max" limits for the joints, in my case is Slider (distance) and Revolute (Angel), the joints will crash, and throw the error as below:
`20:57:55  4453.25 <Assembly> AssemblyObject.cpp(171): Solve failed: invalid vector subscript`

I was still able to drag the part and create the movements, but the error message kept appearing constantly. Additionally, the parent assembly containing this assembly also crashes. You can see the behavior in the video below. Once I unchecked the Min and Max limits, the problem disappeared.
 
Below is the assembly file, please try whether you can reproduce this phenomenon or it just happens to me. Please open the Assy_Sliding file

[Assy_Sliding.zip](https://github.com/user-attachments/files/22311279/Assy_Sliding.zip)

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/02 21:58:19
Build type: Release
Branch: main
Hash: 04e085cba3b43d1e09aedd4445824a6eb46e45b6
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Object center
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.047/1.5
Installed mods: 
  * addFC 1.9.2
  * AddonManager 2025.8.27
  * Cables 0.2.1
  * CurvedShapes 1.0.13
  * Curves 0.6.70
  * fasteners 0.5.41
  * freecad.gears 1.3.0
  * lattice2 1.0.0
  * Nodes 0.1.36
  * OpenTheme 2025.5.20
  * Quetzal 1.5.6
  * SearchBar 1.5.2.1
  * sheetmetal 0.7.52
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-09-15:
I confirm. As soon as one limit is enabled, when you start dragging you get a AssemblyObject.cpp(171): Solve failed: invalid vector subscript.

PaddleStroke, 2025-09-15:
cannot reproduce in 42475 
can reproduce in 43302

PaddleStroke, 2025-09-15:
I think this is very likely due to https://github.com/FreeCAD/OndselSolver/pull/4

@aiksiongkoh the problem here, is that assemblyObject::solve() fails (runKinematic). When the you drag (runDragStep) it works OK.

Conscientiia, 2025-09-19:
Hi @PaddleStroke, based on the WIP blog of the latest release, I see it mentions you have fixed the issue relating to the crash when enabling the limits, not sure if it relates to this issue.

I also tried the latest release as well, and still get that error  "11:24:50  1368.48 <Assembly> AssemblyObject.cpp(171): Solve failed: invalid vector subscript" when dragging the joints, FC didn't crash, and the assembly still be able to move.

In case you guys want a file for testing out or reproduce, please see below. I downloaded it from GrabCAD, and just added some joints for validating purpose.

[Main Landing Gear.zip](https://github.com/user-attachments/files/22419223/Main.Landing.Gear.zip)

aiksiongkoh, 2025-09-28:
Working on it. Thanks.

closed 2025-09-29 by chennes