FreeCAD/FreeCAD #23954: Core: Dependency Graph in SVG

leoheck, 2025-09-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The dependency graph is currently given in a static picture in a tab inside Fc.
But it could be generated in SVG and loaded with the browser by default. 

This way, users could analyze things quickly, especially on complex designs.

[image]rr

Alternatively, settings could have a place with the used Graphviz command, so the user could tweak it directly.

_Yes, I am aware we can export the Graphviz file. This one above was generated using it._

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/12 03:41:38
Build type: Release
Branch: main
Hash: 952bff5b63a62b55a7ea1623edfb39e90dee2d4c
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
Installed mods: 
  * Assembly4.1 0.60.2.1
  * Cables 0.2.1
  * CurvesWB 0.6.64
  * FreeCAD-render 2024.12.15
  * sheetmetal 0.7.52
  * DynamicData 2.75.0
  * Curves 0.6.70
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2025.5.20
  * fasteners 0.5.41
  * Silk 0.2.0
  * osh-autodoc-workbench 0.2.3
  * CurvedShapes 1.0.13
  * freecad-povray-render
  * OpticsWorkbench 1.2.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.7.6
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-09-16:
https://wiki.freecad.org/Std_DependencyGraph#Save

leoheck, 2025-09-17:
@Roy-043 

I was not able to produce a usable file (using `Ctrl+s` in this case if it makesa a difference).

Saving it.

[image]

It looks like an SVG indeed.

```
~ …
✦ ➜ file /tmp/project.svg 
/tmp/project.svg: SVG Scalable Vector Graphics image

~ …
✦ ➜ du -sh /tmp/project.svg 
524K	/tmp/project.svg

~ …
✦ ➜ cat /tmp/project.svg | head -n 10
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.1//EN"
 "http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd">
<!-- Generated by graphviz version 2.42.4 (0)
 -->
<!-- Pages: 1 -->
<svg width="31567pt" height="2307pt"
 viewBox="0.00 0.00 31567.00 2306.50" xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink">
<g id="graph0" class="graph" transform="scale(1 1) rotate(0) translate(4 2302.5)">
<polygon fill="white" stroke="transparent" points="-4,4 -4,-2302.5 31563,-2302.5 31563,4 -4,4"/>
```

But when trying to open it... with the default tool for that (gnome image viewer) it fails.

[image]

Inkscape and google-chrome load it tho.

[image]

Maybe the file is too big?

[image]

leoheck, 2025-09-18:
sure, I was just checking why I opened the issue in the first place.