FreeCAD/FreeCAD #23959: Working Varset adjusted variables in sketches,  make zero changes to 3d solid model unless saved and reopened

tech4teachers, 2025-09-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Just a simple trapezoidal solid, one rectangle lofted to another on a reference plane both rectangles were controlled via a varset to stay proportional to each other so the upper one always remained smaller ... thus a trapezoidal solid.  Variables were able to easily and quickly change the two rectangles live in mere moments, but once lofted the 3d solid lost all influence via the varset.  Yet... after saving and reopening the variables again quickly and easily changed the solid ... the expected behavior... in mere moments.

Literally just a pair of rectangles base depth and length and a height between them..... nothing fancy, set the width of the one on top  to always be a var's value of smaller inset size ... similar to a wall thickness  but resulted in a tapered pad. 100% working vars as controls... until it became a 3d solid.

... and yes I tried forced update refreshes all the way up and down the line  prior to the save and re-open 
variables changed... sketches changed.... model just sat there ... no change

Saving and reopening ... insta-solved it.
Not anxious about it... just reporting in case others have a similar issue.
Don't need a flame war of complain-y ubergeeks slagging me off about how I report this.
We aren't having a long drawn out keyboard conversation, this is a one and done report. 

### Full version info

```shell
1.0.2 flatpak in Linux Mint 22.1 Cinnamon on a Lenovo Flex 5
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

depthoffocus, 2025-09-15:
There's a fair bit of missing detail here if you'd like the report to be useful -- could you upload the file that is not working?

luzpaz, 2025-09-16:
Thanks for reporting, tbh this ticket reads like a film noir dream journal. Super hard to follow. 

Also VarSets have a lot more fixes and support in the development build. Any chance you can reproduce on the dev build ? https://github.com/FreeCAD/FreeCAD/releases

tech4teachers, 2025-09-16:
OS: KDE Flatpak runtime (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.2.39319 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 256fc7e)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.12.10, Qt 6.9.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/

tech4teachers, 2025-09-16:
I'll be testing; 
[2025.08.05](https://github.com/FreeCAD/FreeCAD/releases/tag/weekly-2025.08.05)
FreeCAD_weekly-2025.08.05-Linux-x86_64-py311.AppImage
SHA sha256:f810d8d87d34d41d84597390a654dc5333917ad507610bb75fbafac849453896

from just before my release 
and the one just after
[Development Build weekly-2025.08.07](https://github.com/FreeCAD/FreeCAD/releases/tag/weekly-2025.08.07)
FreeCAD_weekly-2025.08.07-Linux-x86_64-py311.AppImage 
SHA sha256:447853d7b077f37beb0e3986b8bd3f02d1fcceee80909ef0723f2efa39011475

depthoffocus, 2025-09-16:
> Asking for the file without suggesting where it can be placed since we all know it is not directly allowed here is a pithy comment at best.

People usually simply zip them up.

tech4teachers, 2025-09-16:
[Paarametric Trapazoidal Solid.FCStd.zip](https://github.com/user-attachments/files/22368124/Paarametric.Trapazoidal.Solid.FCStd.zip)

tech4teachers, 2025-09-20:
**Update, similar but not identical problem with varsets  requiring a restart happened again last night** and I confirmed it again today.
This time I had to restart because while re-using one of the already established working varset calculations 
VarSet.baseDepth / 2 - VarSet.wallThickness  

... in another location ( where I needed the identical measurement
the OK to accept button was greyed out in both the basic EXPRESSION EDITOR ....and when I tick the box to "Show Variable Sets"... in both instances the formula .... the same one literally working elsewhere ... would paste in and then never accept.. OK was not select-able and despite the formula above .. always reverted to the random dimension it was from rough sketch work.

... and yet.... 
Saving and Restarting allowed the formula to work again 
BUT ...Only in the Expression Editor .... never in the editor when the " Show Variable Sets" tick box was checked ....
so .... progress .... sort of??

... at least I was able to use the dimension and change it automatically by adjusting the variables globally .... but the inability to  tick the box for "Show Variable Sets" is truly an odd one.... that is usually where I build a variable controlled expression the first time by picking the variables I want to use in the formula by mouse click.

.... and ... yes... I did try manually typing them ... without the menus selection method ... same problem ... easy to try with my 4 basic variables... still didn't work though.

 ( this is back in my original Flatpack 1.0.2... need to try this in the preceding and following dev builds) 

I will also try to replicate this in the Windows 1,0.2 release ... I suspect this may be a more reproducible variant of the restart-to-use-variable-sets issue as I am now calling it

tech4teachers, 2025-09-22:
Irreproducible in Windows 10,  on any version ... shifting focus back to dev builds and differences between flatpak, and appimage variants in Mint 22.

tech4teachers, 2025-12-14:
Thanks for confirming I'm not the only one with this issue, seemed to crop up more in the Flatpack version, than in the appimage for me but, like you, I need to make a hyper-simplified test object for checking different workflows and how they solve or re-manifest the problem.
I was thinking of making a short video and tagging Mango Jelly on this to see if he can help.... but I can also post it for the millennials who can't read. 
I am back on a windows machine for the holidays, and getting a teensy mount of time off I might get to spend a few hours replicating this on that installed variant..... thanks for the reminder to do this!    :  )

eblanshey, 2025-12-17:
Hi @luzpaz, sorry for bumping during a busy period, but who can we ping to look into this bug and set it as a 1.1 release blocker? I was close to ordering expensive CNC parts that had incorrect dimensions due to this bug. I can't overstate enough the severity of modifying variables and not having confidence that all expressions were updated in the 3d view.

eblanshey, 2025-12-19:
Ah, bummer. Thank you for checking, @ipatch.