FreeCAD/FreeCAD #23971: Core: connect failed: No such file or directory

furgo16, 2025-09-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When starting FreeCAD on the command line, this warning is shown:

```
connect failed: No such file or directory
```

This forum thread: https://forum.freecad.org/viewtopic.php?t=39520 claims that:
- it comes from libspnav
- the issue was fixed upstream 5 years ago

However, it does not seem to be fixed. I've been seeing it at least since FreeCAD 0.21. 

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.43489 (Git) Snap 1722
Build date: 2025/09/15 13:29:55
Build type: Release
Branch: main
Hash: eec6f4ec0d32ec47c6aefc0acec6964df336fcac
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.2, OCC 7.7.1
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Telemetry 1.0.2
  * Reporting
  * Launcher
  * Telemetry-backup-before-zip-migration 1.0.1 (Disabled)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

chennes, 2026-09-11:
If you check the version numbers here, you'll see that this output is *present* in libspnav 0.2.3, which is what that core22 snap included, which is why the upstream fix hasn't reached you yet. @luzpaz is right that @wwmayer did indeed put a "silencer" in this from the FreeCAD side, in cfe94b00d3 but it turned out to break something else, so was removed again in 7b377a216b. Considering that spnav 0.2.3 is now something like six years old, and every subsequent version fixes this in the upstream, I don't think it's worth keeping this bug open here.

luzpaz, 2026-09-11:
Agree with chennes, this is specifically related to Snap packaging because it's running an old version of libspnav.
Closing this per chennes' recommendation.

closed 2026-09-11 by luzpaz

furgo16, 2026-09-12:
Note we're on core24 in the FreeCAD snap now. While the message did appear there at some point (with `libspnav0` 1.1-2), I can no longer reproduce this, so it's fine to close this issue. Thanks a lot for following up.