FreeCAD/FreeCAD #23973: Snap symbol size increase in Draft Workbench (FreeCAD 1.0.2)

Ernurcum, 2025-09-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hello,

I am using FreeCAD 1.0.2 and working in the Draft Workbench.
I want to increase the on-screen size of the snap symbols (the little markers that appear when snapping to endpoints, midpoints, etc.).

So far, I tried changing:

snapRange parameter

snapStyle parameter

Draft preferences in Edit → Preferences → Draft

But these only affect the snap distance or style, not the symbol size.
Even looking into gui_snapper.py → showradius() and adjusting didn’t seem to work.

Is there any way to increase the snap symbol size on screen in FreeCAD 1.0.2?
Maybe via a parameter, Python console, or a small code modification?

### Full version info

```shell
FreeCAD 1.0.2 
Windows 11 
Lenovo
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-09-16:
See: https://wiki.freecad.org/Draft_Snap#Preferences

Ernurcum, 2025-09-17:
The size of the snap symbols depends on: Edit → Preferences → Display → 3D View → Marker size. [introduced in 1.0](https://wiki.freecad.org/Release_notes_1.0)
not the actual point size i want to increase showing symbol of snap (like perpendecular sign _I_ on screen)

Ernurcum, 2025-09-17:
like in autocad 

[image]