FreeCAD/FreeCAD #23975: Crash whenever multi-transform linear pattern iterations reaches 10

hololight, 2025-09-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

While trying to follow the MangoJelly tutorial (https://youtu.be/YWgCCgpxNCQ?si=QBPj4KjCn1Twhw3d), when I try to create a linear pattern of a subtractive hole, I can increase the occurrences up to 9, and it works correctly. Once I add one more, trying for 10, FreeCAD immediately crashes. I have repeated the issue multiple times with the exact same behavior. The same thing was happening with the weekly version from 2025.08.20.

### Full version info

```shell
OS: macOS 15.6.1
Architecture: arm64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/12 03:41:38
Build type: Release
Branch: main
Hash: 952bff5b63a62b55a7ea1623edfb39e90dee2d4c
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/151.5/2
Installed mods: 
  * Gridfinity 0.11.9
  * FusedFilamentDesign 0.25.250
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

hololight, 2025-09-17:
I just tried in 1.0, and it seems to work fine (only tested to 10 occurrences, where it is crashing the nightlies).

[Drain.2.FCStd.zip](https://github.com/user-attachments/files/22390198/Drain.2.FCStd.zip)

maxwxyz, 2025-09-17:
@PaddleStroke is this related to pattern changes?

hololight, 2025-09-17:
I dont know if this helps, but If I create a pad from a simple square, then create my hole and linear pattern it, it works fine. If I create a rectangle and a revolve, and again a hole and pattern it, it again works fine. It is something about the particular thing I've created following that tutorial and I've uploaded in the 'drain' file...

PaddleStroke, 2025-09-18:
[image]

If you change the sketch from 40mm to 38mm, then it doesn't crash anymore. So it feels very unlikely that it has something to do with recent changes.

Also it doesn't need to be multi-transform. A simple linear transform also crashes at 10.

People on linux do you have some logs to share please?

@chennes Please rename it to remove the multi-transform from the title as it is just on linear pattern. It would be useful if I had write access so that I could edit issues when needed.

PaddleStroke, 2025-09-18:
additional information 2 : 
- If you change occurence to 10 in the property view, the crash also happens.
- If you change the shape of the plate, for example I changed from 10 to 20 the dimension on top left :

[image]

Then it does not crash.

Long story short, this is very likely a TNP bug.

hololight, 2025-09-19:
Also I just discovered... if you change the linear pattern Mode to 'Spacing', instead of 'Extent', it will correctly make it to 10 Occurances...

chennes, 2025-09-21:
I just tried this and did not get the failure -- maybe I didn't follow the steps exactly? I've attached my FCStd file with ten occurrences of the subtractive pattern.

[Drain.2.with_pattern_x10.FCStd.zip](https://github.com/user-attachments/files/22454516/Drain.2.with_pattern_x10.FCStd.zip)

PaddleStroke, 2025-09-22:
https://github.com/user-attachments/assets/430aa9be-30d5-450b-b2eb-67fd4dcad30e

@chennes here's reproduction steps video

pieterhijma, 2025-12-09:
> I just tried in 1.0, and it seems to work fine (only tested to 10 occurrences, where it is crashing the nightlies).
> 
> [Drain.2.FCStd.zip](https://github.com/user-attachments/files/22390198/Drain.2.FCStd.zip)

Could you please tell me which OCCT version your 1.0 is using? Thanks in advance.

maxwxyz, 2025-12-09:
Therefore it cannot be a blocker too, removing the tag. Let me know otherwise.

3x380V, 2026-04-25:
FYI, @wwmayer's fix:  https://codeberg.org/wwmayer/FreeCAD11/commit/f33d116991542f3cacc9114cb2e51f47ea650b1d