FreeCAD/FreeCAD #23979: python import issues with FreeCAD_weekly-2025.09.03-Linux-x86_64-py311 onwards

APEbbers, 2025-09-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

From FreeCAD_weekly-2025.09.03-Linux-x86_64-py311onwards I have problems with my workbenches.
It seems that codes like "print(App.Version()", print() etc are no longer working for some reason.
Code like "os.path.isfile(.." is also not working. This results in issues with the Ribbon UI.

I have no problems with older versions. Is there something changed in the way FreeCAD should be imported into the python code that i'm unaware of??

Or has this perhaps to do with Wayland?

Kind regards

Paul Ebbers

### Full version info

```shell
OS: EndeavourOS (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: SiemensNX/Free Turntable/Object center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/109.57/1
Installed mods: 
  * manifest.json
  * SearchBar 1.6.0
  * AddonManager 2025.8.27
  * ProDarkThemePreferencePack 1.0.0
  * MakerWorkbench 1.0.1
  * BillOfMaterials 1.1.0.1
  * A2plus 0.4.68
  * SaveAndRestore 0.2.1
  * Assembly3 0.12.2
  * FreeCAD-Ribbon 1.10.0
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

I'm the developer a few addons. The most important are: 
- Ribbon UI
- SearchBar
- SaveAndRestore
- BillOfMaterials.

I have the biggest issue with the Ribbon UI, which depends on a correct import of the FreeCAD libraries.

Arch Linux hadn´t not yet updated their link to the AppImage until recently. This is why I'm a bit late with this issue.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-09-17:
@chennes any idea?

Syres916, 2025-09-17:
For me this is extremely serious and got to be _Blocker_ status. Even if I create a folder under `/home/username/.local/share/FreeCAD/Mod` , call it `ChangeAppFont` for example and create a `InitGui.py` populated with known good code. On starting a recently compiled FreeCAD it doesn't even recognise the `ChangeAppFont` as an Addon to be actioned, completely missing from the Report View output.

Syres916, 2025-09-17:
The last good build I have is:

```
OS: Debian GNU/Linux 13 (trixie) (X-Cinnamon/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.1.0dev.43173 (Git)
Build date: 2025/08/31 19:53:51
Build type: Release
Branch: (no branch, bisect started on main)
Hash: 25d1a51fb401feb55dcbb0eae8f55fc0add4a35d
Python 3.13.5, Qt 6.8.2, Coin 4.0.3, Vtk 9.3.0, boost 1_83, Eigen3 3.4.0, PySide 6.8.2.1
shiboken 6.8.2.1, SMESH 7.7.1.0, xerces-c 3.2.4, IfcOpenShell 0.8.3.post2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/96.0757/1
```

chennes, 2025-09-18:
I haven't looked at the code for your Addons, but the idea is that this should be invisible to Addons. Can you tell me more about what was going wrong?

APEbbers, 2025-09-18:
The first time the Ribbon is installed, it creates a json file called "RibbonStructure.json", which contains all the data for the buttons and layout of the Ribbon. After that the location of that file is stored as a parameter. The next time FreeCAD starts, the ribbon uses the location stored in parameters to load the json file again.  My code checks if the file is present in the stored location. if it is not, it recreates the file again in the folder of the addon. But because the parameter was pointing to the old location, the Ribbon never saw the new created file, which is present at the new location.

So now I've implemented a fix that checks which version of FreeCAD is used. If it is FreeCAD 1.1.0 git version 1.4555 or higher,
the ribbon looks for the parameter "AddonDir" in parameters. if it is not present, it means that the parameters needs to be updated to the new location. It then also creates the parameter "AddonDir", so that knows that the parameters are already updated next time FreeCAD starts.

The reason that I stored the location of the "RibbonStructure.json" file as a parameter, is to provide other creators of addons the possibility to create a custom layout of the Ribbon for their addons. They can read the file, updated it and save it.
It is also possible to store the file at a completely different location by changing the parameter manually. This could be interesting for companies in case they want to use the Ribbon with a company layout for example.

I hope that it makes sense. Perhaps i made it a bit too complicated. 

kind regards

chennes, 2025-09-19:
I think in your use case the critical thing would be to store the path *relative* to `userAppDataDir` (or `configDir` if that makes more sense for where you expect it to be). Then after future migrations (for example as soon as we increment the version to 1.2 after the 1.1 branch is made) your addon would not need to worry about it.

I'm not sure how I could write anything that would be able to programmatically detect if an addon is doing what yours is but I am open to ideas. Maybe @mnesarco has some ideas?