FreeCAD/FreeCAD #23988: TechDraw: Arbitrary movement of the projection

Pendelf, 2025-09-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Arbitrary movement of the projection during selection
In version 1.0, there is no such problem.

https://github.com/user-attachments/assets/0fd99eba-a72b-4e07-9eb0-031064fe119d

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * addFC 1.9.1
  * CurvedShapes 1.0.13
  * Curves 0.6.70
  * fasteners 0.5.41
  * PieMenu 1.12.0
  * sheetmetal 0.7.24
  * StructureTools 0.0.1
  * SvgWorkbench 1.0.0.dev6
  * Telemetry 1.0.2
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-09-17:
@ryankembrey FYI

ryankembrey, 2025-09-17:
Looks to me like when the root view and child view are both selected, root view moves x/y drag distance and the child view moves 2x/2y distance. So the drag is being added to the child twice