FreeCAD/FreeCAD #23998: TechDraw: The view frame does not reset with a click

Pendelf, 2025-09-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

If you select a frame and the element inside it with Ctrl, the selection of the frame won't be cleared by clicking on an empty space

https://github.com/user-attachments/assets/635f0b6a-728b-49c1-9894-6461444e4ee2

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * addFC 1.9.1
  * CurvedShapes 1.0.13
  * Curves 0.6.70
  * fasteners 0.5.41
  * PieMenu 1.12.0
  * sheetmetal 0.7.24
  * StructureTools 0.0.1
  * SvgWorkbench 1.0.0.dev6
  * Telemetry 1.0.2
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

WandererFan, 2025-09-17:
Recently merged PR:  TechDraw: Fix selections not clearing #23587

ryankembrey, 2025-09-18:
Note that this bug is a seperate issue and was not solved in https://github.com/FreeCAD/FreeCAD/pull/23587

ryankembrey, 2025-09-18:
Fixed with #24021

PR #24021 (2025-09-29): TechDraw: Fix selections not clearing when subelements are also selected
 Fix selections not clearing when subelements are also selected.

## Issues
Fix #23998

## Demo of Fix

https://github.com/user-attachments/assets/70684c78-b565-4dcf-93ef-0e97eef0f5ae