FreeCAD/FreeCAD #24005: Allow External reference property to be changed in the Property view.

macdroid53, 2025-09-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In the past you could correct missing external refs by changing them  in the Property view. This has been set to read only since the release of 1.0.

With this forced to RO, there are few ways to fix TNP when it occurs.

Can this be set to at least allow the user to override RO?

### Full version info

```shell
OS: Debian GNU/Linux 13 (trixie) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.43510 (Git)
Build date: 2025/09/16 20:38:23
Build type: Unknown
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Object center
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/93.6146/1
Installed mods: 
  * fasteners 0.5.41
  * sheetmetal 0.7.53
  * lattice2 1.0.0
  * OpenTheme 2025.5.20
  * OpticsWorkbench 1.2.2
  * CurvedShapes 1.0.13
  * Alternate_OpenSCAD 1.0.0
  * Curves 0.6.70
  * Defeaturing 1.2.9
  * Silk 0.2.0
  * Channels 0.1.0.dev3
  * kicadStepUpMod 11.7.6
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

macdroid53, 2025-09-18:
Any file with a PD Body, 2 sketches, and an external reference from one to the other.. The field is greyed out.

[image]

macdroid53, 2025-09-19:
@maxwxyz  This is not a bug or a feature.

It is a regression.

In the past, you could use this to manually correct TNP, now you can't.

maxwxyz, 2025-09-19:
@drwho495 any idea about that?
@PaddleStroke was this changed in sketcher with the new external geometries?

macdroid53, 2025-09-19:
I think Brad changed it during the TNP mitigation merge.

chennes, 2025-09-22:
It is simple enough to activate that field for editing, but the reason it's disabled is because editing it doesn't work. I poked at it for a few minutes today and couldn't achieve anything productive via a direct edit, there's too much going on behind-the-scenes with the object names. @PaddleStroke might have some insight as to whether there's a way to make this work, I think external geometry is sort of his specialty.

FEA-eng, 2025-11-23:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20251119 (Git shallow)
Build date: 2025/11/19 23:28:17
Build type: Release
Branch: main
Hash: deab0abf8e61d69f2208b794f64ac56428467fef
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

luzpaz, 2025-12-25:
Added 'Regression' to the title.