FreeCAD/FreeCAD #24008: FreeCAD crashes after restoring from sleep

cropduster32, 2025-09-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

FreeCAD crashes on saving new project

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/Fusion
Installed mods: 
  * ThreadProfile 1.96.0
  * lattice2 1.0.0
  * Curves 0.6.64
  * A2plus 0.4.68
  * CurvedShapes 1.0.13
  * Pyramids-and-Polyhedrons
```

### Subproject(s) affected?

None

### Anything else?

After logging in to the laptop after restoring from sleep I restored FreeCAD from a previous instance. When creating a new project and immediately saving the new one FreeCAD crashes with no messages or displayed error value. I cannot find a file that may have been created at crash.

After running FreeCAD afresh then loading any previously opened projects, then creating a new project and saving as above, it does not crash. It seems only to crash after restoring from sleep! 

The laptop is a DELL 7620

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

macdroid53, 2025-09-17:
I haven't experienced this recently. But I have experienced it with the weekly appimage on Unbuntu.

nenadandric, 2026-01-18:
No issues with Version: 1.1.0rc2.20260108 (Git shallow) AppImage

closed 2026-04-19 by maxwxyz