FreeCAD/FreeCAD #24010: Techdraw: it is impossible to access horizontal scroll bar on the page while report view on overlay

ErtugrulEge, 2025-09-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings to everyone!
In techdraw page you cannot access to horizontal sscroll bar because when mouse hover above report view, report view opens. Of course you can close report view it seems to me a thing that should be fixed

https://github.com/user-attachments/assets/13d31471-4c45-479d-b00b-fe3fc1921876

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Free Turntable/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/127.628/1
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-11-16:
@ErtugrulEge can you still reproduce on latest dev ?

Edit: this explanation from realthunder may be the reason for this phenomena  
* https://github.com/FreeCAD/FreeCAD/issues/11637#issuecomment-1854908810

ErtugrulEge, 2026-10-15:
Still relevant 

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 27.1.0dev.20261013 (Git shallow)
Build date: 2026/10/13 19:55:30
Build type: Release
Branch: (HEAD detached at 50daee777)
Hash: 50daee777321b5085accd1f57b57f1a7c1781cae
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Turkish/Türkiye (tr_TR)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 597.06/NVIDIA Corporation/Quadro RTX 4000/PCIe/SSE2
Installed mods:
  * Curved Shapes 1.0.15
  * Curves workbench 0.6.81

```