FreeCAD/FreeCAD #24013: real-time draft becomes slow

gianlucafiore, 2025-09-17:
I reiterate this issue, when starting a blank project from the first opening of the software, I can draw any line or wall bim_wall without the slightest lag or delay (that manual layout of that first entity has excellent performance) when finishing drawing that first entity, a kind of lag occurs in the entire software of about 2 seconds, and when returning everything that is drawn, it is done slowly and lagged (to a very small extent) but it is felt

> I'm experiencing interface slowdown again after creating the first wall, or drawing the first line since starting the software.
> 
> [2025-08-14-19-20-33_4RAneAoR.mp4.zip](https://github.com/user-attachments/files/21781301/2025-08-14-19-20-33_4RAneAoR.mp4.zip) **delete.zip extension to reproduce**
> 
> Steps to reproduce:
> - Open FreeCAD from scratch
> - Start with an empty file
> - Draw the first wall or line or whatever (it does very well with the preview)
> - Then there's a lag of 1 or 2 seconds
> - Then try to draw the next wall or line and the preview is very slow for all the rest of the work until you close the program
> 
> [image]
> 
> And GPU is on 
> 
> @Syres916  

 _Originally posted by @gianlucafiore in [#22390](https://github.com/FreeCAD/FreeCAD/issues/22390#issuecomment-3190051097)_

Roy-043, 2025-09-17:
Why do you close a previous issue to only report the same again?

semhustej, 2025-09-18:
I can reproduce the issue. I recently (a month ago ?) started to get delays even when starting BIM/Draft commands after FreeCAD is started. I am not able to reproduce the issue 100% time though, and I did not find out what influences the lag.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.2

```

semhustej, 2025-09-18:
Note: similar behavior reported by another user here: https://github.com/FreeCAD/FreeCAD/issues/17971#issuecomment-3289706164

Roy-043, 2025-09-18:
Are you using the mentioned versions on the same OS and hardware?

gianlucafiore, 2025-09-18:
Yes, exactly, same pc

OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Spanish/Argentina (es_AR)
Navigation Style/Orbit Style/Rotation Mode: Revit/Free Turntable/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/141.585/1
Installed mods: 
  * FreeCAD_SketchArch
  * Curves 0.6.70
  * Reporting
  * Beltrami 1.1.0
  * A2plus 0.4.68
  * SaveAndRestore 0.2.1
  * SearchBar 1.4.2
  * FreeCAD-Ribbon 1.9.1.14

Roy-043, 2025-09-18:
Can you check in Safe Mode?
wiki.freecad.org/Std_RestartInSafeMode

gianlucafiore, 2025-09-18:
Can you see what I share below in the videos?
[Drive folder - test videos](https://drive.google.com/drive/folders/1n2xOsyd3ZMWV-ZImiCOaoEx17-MEkeYS?usp=sharing)

kaiwas, 2025-09-19:
Maybe these problems are related?

https://github.com/FreeCAD/FreeCAD/issues/24032

PR #24081 (2025-09-22): Draft: fix snapper lagging
Fixes #24013.

If `update=True` the `WorkingPlane.get_working_plane()` function is too slow to be called repeatedly via `_get_wp`.