FreeCAD/FreeCAD #24018: Core: Object Tree Should Not Auto-Scroll to Errors or Active Features

leoheck, 2025-09-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Can we make the Object Tree stop automatically scrolling and focusing on a problematic sketch or feature?

This is extremely annoying. The Tree view should be static. It should not scroll or change state without the user's intervention.

I am working on other tasks and cannot keep scrolling up every time something changes.

If this behavior is by design, I would ask for an option to disable it—perhaps when Skip Recomputes is enabled.

In a large and complex project, this is counterproductive. Multiple things can break at the same time, but the system shouldn’t constantly drag the user’s attention to each one. We already have a report view and other places for that.

[image]

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
Installed mods: 
  * Assembly4.1 0.60.2.1
  * Cables 0.2.1
  * CurvesWB 0.6.64
  * FreeCAD-render 2024.12.15
  * sheetmetal 0.7.52
  * DynamicData 2.75.0
  * Curves 0.6.70
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2025.5.20
  * fasteners 0.5.41
  * Silk 0.2.0
  * osh-autodoc-workbench 0.2.3
  * CurvedShapes 1.0.13
  * freecad-povray-render
  * OpticsWorkbench 1.2.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.7.6
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

dekay, 2025-09-18:
I have seen this same behavior. I find it very frustrating as well, especially on a large project.

FlachyJoe, 2025-09-18:
+1
But to increase their visibility, Warnings and Errors could be shown in a separate view. Perhaps at the bottom of the tree.

leoheck, 2025-09-22:
This feels more like a bug than a feature.
Why does the Object Tree behave like that?

It always focuses on the last object that broke.
But I don’t think anyone works from the bottom up in the tree.
Most people fix issues from the top down, especially since that’s the natural order in which objects appear and the way they’re related.