FreeCAD/FreeCAD #24027: BIM: BIM_Setup problematic AcadEref text font set as default

semhustej, 2025-09-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

`AcadEref` is set as a default text font in the BIM_Setup. This means it is recommended by FreeCAD to use with BIM workbench.

The problem is that this text font is broken in FreeCAD - both the vertical alignment and letter spacing. Example of this font used in a dimension:

[image]

Steps to reproduce:

- open FreeCAD
- open `BIM_Setup`
- choose `OK` to accept the defaults. `AcadEref` font is selected
- create Draft_Dimension
- the text is completely non-usable as the picture above shows

[FC-AcadEref-text.FCStd.zip](https://github.com/user-attachments/files/22402693/FC-AcadEref-text.FCStd.zip)

**Proposed solution:**
Use `Arial` as a default text font for all pre-set options in the `BIM_Setup` dialogue

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
```

### Subproject(s) affected?

BIM

### Anything else?

This is quite a serious problem which should be treated as a bug and and I think it should be fixed for the 1.1 release, because it is hitting every new user who accepts the default BIM settings and it is not very easy to troubleshoot it.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-09-18:
Which font do you get when you paste this in the Python Console:
```python
from PySide import QtGui
font = QtGui.QFont()
font.setStyleHint(QtGui.QFont.StyleHint.SansSerif)
value = font.defaultFamily()
print(value)
```

semhustej, 2025-09-18:
> from PySide import QtGui
> font = QtGui.QFont()
> font.setStyleHint(QtGui.QFont.StyleHint.SansSerif)
> value = font.defaultFamily()
> print(value)

I get this:

`Arial`

semhustej, 2025-09-18:
Additional information:

When testing in Safe Mode, the proposed font in BIM_Setup is Arial.

When I did a test (non safe mode) on a different windows machine woth FC 1.0.2 the proposed font in `BIM_Setup` was `MS Shell Dlg 2` , which I never used.

Roy-043, 2025-09-18:
I suspect this then due to a faulty parameter, perhaps inherited from a previous version.
Please check:
Tools -> Edit Parameters -> Preferences -> Mod -> Draft -> textfont