FreeCAD/FreeCAD #24036: Renaming doesn't cascade through object tree

Lucretia, 2025-09-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Create an object, then later rename something something else is based on, then try to delete something, the tree gets out of sync and cannot continue, you get errors and either have to revert or recreate the object, a real pain.

### Full version info

```shell
1.0.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-09-18:
Please give full version info and supply a reproducible scenario.

luzpaz, 2025-11-25:
@Lucretia can you reproduce on the weekly development build ? Please test and report back. Thanks!
https://github.com/FreeCAD/FreeCAD/releases

Note: please always paste your full [About](https://wiki.freecad.org/About) info when reporting tickets. It really helps reduce a lot of back and forth. thanks!

Edit:

> one small trees there doesn't seem to be an issue,

What does this mean ?

FEA-eng, 2026-10-26:
See [this](https://github.com/FreeCAD/FreeCAD/issues/24036#issuecomment-3950074276)