FreeCAD/FreeCAD #24037: PartDesign Trim causing Crash

InstantMuffin, 2025-09-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

See video and test file.

[CrashTest.FCStd.zip](https://github.com/user-attachments/files/22410308/CrashTest.FCStd.zip)

[image]

### Full version info

```shell
Latest FreeCAD weekly at the time of writing. XUbuntu x64
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

InstantMuffin, 2025-09-18:
Output when started from terminal:

`Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45330) [0x7e584d645330]
#1  0x7e57ad12b085 in Sketcher::SketchObject::trim(int, Base::Vector3<double> const&) from /tmp/.mount_FreeCAGOJpiF/usr/lib/Sketcher.so+0xab5
#2  0x7e57ad252c5e in Sketcher::SketchObjectPy::trim(_object*) from /tmp/.mount_FreeCAGOJpiF/usr/lib/Sketcher.so+0x7e
#3  0x7e57ad243e48 in Sketcher::SketchObjectPy::staticCallback_trim(_object*, _object*) from /tmp/.mount_FreeCAGOJpiF/usr/lib/Sketcher.so+0x48
#4  /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libpython3.11.so.1.0(+0x1faeb8) [0x7e584fffaeb8]
#5  /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x2b4) [0x7e584ffd9c04]
#6  /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6e7) [0x7e584ffe53c7]
#7  /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libpython3.11.so.1.0(+0x29ef1d) [0x7e585009ef1d]
#8  /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7e585009e914]
#9  /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libpython3.11.so.1.0(+0x2bced3) [0x7e58500bced3]
#10  /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libpython3.11.so.1.0(+0x2b8aea) [0x7e58500b8aea]
#11  /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7e58500acfe9]
#12  0x7e585117fa01 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libFreeCADBase.so+0x71
#13  0x7e58539bac21 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libFreeCADGui.so+0xa1
#14  0x7e58539baaaf in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libFreeCADGui.so+0xdf
#15  0x7e57ac770107 in void Gui::cmdAppObjectArgs<int&, double&, double&>(App::DocumentObject const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int&, double&, double&) from /tmp/.mount_FreeCAGOJpiF/usr/lib/SketcherGui.so+0x207
#16  /tmp/.mount_FreeCAGOJpiF/usr/lib/SketcherGui.so(+0x56fdf0) [0x7e57ac76fdf0]
#17  /tmp/.mount_FreeCAGOJpiF/usr/lib/SketcherGui.so(+0x56f97a) [0x7e57ac76f97a]
#18  0x7e57ac89536c in SketcherGui::ViewProviderSketch::mouseButtonPressed(int, bool, SbVec2s const&, Gui::View3DInventorViewer const*) from /tmp/.mount_FreeCAGOJpiF/usr/lib/SketcherGui.so+0xa6c
#19  0x7e5853de13d8 in Gui::ViewProvider::eventCallback(void*, SoEventCallback*) from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libFreeCADGui.so+0x248
#20  0x7e584f8d346e in SoEventCallback::handleEvent(SoHandleEventAction*) from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libCoin.so.80+0xbe
#21  0x7e584f6d54e8 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libCoin.so.80+0x68
#22  0x7e584f88cb9a in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libCoin.so.80+0xea
#23  0x7e584f8dc348 in SoGroup::doAction(SoAction*) from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libCoin.so.80+0x58
#24  0x7e584f8f713a in SoSeparator::doAction(SoAction*) from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libCoin.so.80+0x2a
#25  0x7e584f6d54e8 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libCoin.so.80+0x68
#26  0x7e584f88cb9a in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libCoin.so.80+0xea
#27  0x7e584f8dc348 in SoGroup::doAction(SoAction*) from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libCoin.so.80+0x58
#28  0x7e584f8f713a in SoSeparator::doAction(SoAction*) from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libCoin.so.80+0x2a
#29  0x7e584f6d54e8 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libCoin.so.80+0x68
#30  0x7e584f6df26c in SoHandleEventAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libCoin.so.80+0xac
#31  0x7e584f6d61ba in SoAction::apply(SoNode*) from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libCoin.so.80+0x2ea
#32  0x7e584f8a8ad2 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libCoin.so.80+0x72
#33  0x7e5853d670f9 in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libFreeCADGui.so+0x19
#34  0x7e5853d6e4a6 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libFreeCADGui.so+0x86
#35  0x7e5853dcb22d in Gui::NavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libFreeCADGui.so+0x4d
#36  0x7e5853dd02f7 in Gui::BlenderNavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libFreeCADGui.so+0x6f7
#37  0x7e5853dcad65 in Gui::NavigationStyle::processEvent(SoEvent const*) from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libFreeCADGui.so+0x95
#38  0x7e5853d8e293 in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libFreeCADGui.so+0xb3
#39  0x7e5853d6138f in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libFreeCADGui.so+0x12f
#40  0x7e584dd5f8de in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libQt6Core.so.6+0x7e
[... cut ...]
#59  0x7e584dd977da in QCoreApplication::exec() from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libQt6Core.so.6+0x9a
#60  0x7e585387c20d in Gui::Application::runApplication() from /tmp/.mount_FreeCAGOJpiF/usr/bin/../lib/libFreeCADGui.so+0xc6d
#61  /tmp/.mount_FreeCAGOJpiF/usr/bin/freecad(+0x9bb5) [0x58a86515fbb5]
#62  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x7e584d62a1ca]
#63  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x7e584d62a28b]
#64  /tmp/.mount_FreeCAGOJpiF/usr/bin/freecad(+0x8cd9) [0x58a86515ecd9]
`

maxwxyz, 2025-09-19:
Please test in the new weekly next Wednesday

InstantMuffin, 2025-09-19:
> Looks like it is the same issue as [#23882](https://github.com/FreeCAD/FreeCAD/issues/23882) ?

I agree. Interesting it didn't come up in the search. I searched for "trim", "trimming" apparently doesn't count. Thank you, good catch!