FreeCAD/FreeCAD #24065: BIM: Arch_Stairs Blondel Ratio property not calculated

semhustej, 2025-09-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

`Blondel Ratio` property is not calculated in `Arch_Stairs` object.

[image]

Steps to reproduce:

- open new file
- create `Arch_Stairs` object
- `Blondel Ratio` parameter is calculated as 0
- I was not able to change that setting different `Riser Height` and `Tread Depth` properties

[image]

I am able to reproduce the issue in FC 1.0.2

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.2
```

### Subproject(s) affected?

BIM

### Anything else?

According to Wikipedia Blondel Ratio should be calculated as:

BR = 2 × riser height + tread depth

https://en.wikipedia.org/wiki/Stairs#Ergonomics_and_building_code_requirements

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

paullee0, 2025-09-20:
Indeed I remember it was not working.

Is it an important information?

paullee0, 2025-09-21:
I seems below properties are not updated also, correct?

1. TreadDepth
2. RiserHeight

paullee0, 2025-09-21:
@semhustej or anybody can test?  Thanks.

[ArchStairs.py.TXT](https://github.com/user-attachments/files/22449543/ArchStairs.py.TXT)

semhustej, 2025-09-21:
> Seems TreadDepth and RiserHeight are updated in code elsewhere.
> 
> Can test this one? Thanks.
> 
> [ArchStairs.py.TXT](https://github.com/user-attachments/files/22449585/ArchStairs.py.TXT)

This last library works on my side. The `Blondel Ratio` is calculated correctly. I also tested using `Riser Height Enforce` and `Tread Depth Enforce` with success.

[image]

semhustej, 2025-09-21:
**With this new library I get one regression:**
When using `Tread Depth Enforce` property with Stair object which was create with a base object, the `Tread Depth Enforce` property value is not copied to `Tread Depth` property as you can see in this picture:

[image]

This issue does not happen with baseless stair object.

paullee0, 2025-09-21:
Thanks @semhustej for the testing, it really helps.

paullee0, 2025-09-21:
> **With this new library I get one regression:** When using `Tread Depth Enforce` property with Stair object which was create with a base object, the `Tread Depth Enforce` property value is not copied to `Tread Depth` property as you can see in this picture:
> 

> This issue does not happen with baseless stair object.

It seems you tested both versions, could you clarify which one works and which one has regression?

Thanks again :)

paullee0, 2025-09-21:
I see, that clarified the issue, no further test needed, thanks.

semhustej, 2025-09-24:
> Further updated, can try testing?
> 
> 
> [ArchStairs.py.TXT](https://github.com/user-attachments/files/22499207/ArchStairs.py.TXT)

I did a quick test with this latest library and Blondel ratio works for all cases I tested.

**I encountered another issue related to units used:**

The Blondel Ratio should use length units (it is just adding and multiplying height and depth). This is currently not happening.

Steps to reproduce:

- open new file
- change unit system to `US Customary` or `Meter Decimal`
- create `Arch_Stairs` object
- the value of `Blondel Ratio` parameter is shown in milimeters (incorrect - it should show value in inches or meters)
- the parameter does not show units (incorrect - it should show `mm`, `m`, `in` etc. depending on the unit system used)

[image]

paullee0, 2025-09-24:
> I did a quick test with this latest library and Blondel ratio works for all cases I tested.
> 
> **I encountered another issue related to units used:**
> 
> The Blondel Ratio should use length units (it is just adding and multiplying height and depth). This is currently not happening.
> 

I see.  The property currently is unitless.  

Agreed to all to change it to Length?

semhustej, 2025-09-24:
> 
> I see. The property currently is unitless.
> 
> Agreed to all to change it to Length?
> 
> EDIT - Please help testing :) [ArchStairs.py.TXT](https://github.com/user-attachments/files/22508545/ArchStairs.py.TXT)

I tested this latest library and the Blondel Ratio shows units according to the chosen unit system.

`Tread Depth` and `Tread Height` are both `Length` properties, so adding them should result in a Length property even though it is named 'Ratio'. The research I did on the internet and also the pop-up tip shows the property should show length units.

From my point of view all issues seem to be fixed now.

paullee0, 2025-09-24:
Any opinion from others?  Thanks

PR #24350 (2025-10-05): [ArchStairs] Fix Blondel Ratio and Winders
Fix #24065
Fix #24051

1. Blondel Ratio is not calculated, this if fixed now.  Blondel Ratio property is changed to Length following wiki's descripition.
2. Winders is not implemented as commented in wiki.  The property is not added at the moment until it is implemented.

## Issues

## Before and After Images
commit 960434f7 (2025-10-05): [ArchStairs] Fix Blondel Ratio and Winders (#24350)

* [ArchStairs] Fix Blondel Ratio and Winders

Fix #24065
Fix #24051

1. Blondel Ratio is not calculated, this if fixed now.  Blondel Ratio property is changed to Length following wiki's descripition.
2. Winders is not implemented as commented in wiki.  The property is not added at the moment until it is implemented.

* [ArchStairs] Fix Blondel Ratio and Winders - Remove old Property

For existing Stairs object, remove old property which is Float and add new propety which is Length.
closed 2025-10-05 by Roy-043