FreeCAD/FreeCAD #24077: Part Design: Cancel an additive shape operation when there is only assembly in TreeView causing crash

Conscientiia, 2025-09-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I think this may somehow relating to https://github.com/FreeCAD/FreeCAD/issues/23675

This time, when you cancel an additive shape operation when there is only assembly in the TreeView (without any bodies) will cause a crash. I know that this is not a standard way to do but instead of crashing, a warning pop up should be better.

Procedures to reproduce this:

1. Create a new FC file
2. Create an assembly
3. Switch to Part Design WB
4. Click to add an additive shape, and then cancel the operation.

[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Object center
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/151.219/1.25
Installed mods: 
  * addFC 1.9.2
  * AddonManager 2025.8.27
  * Cables 0.2.1
  * CurvedShapes 1.0.13
  * Curves 0.6.70
  * fasteners 0.5.42
  * freecad.gears 1.3.0
  * lattice2 1.0.0
  * Nodes 0.1.36
  * OpenTheme 2025.5.20
  * Quetzal 1.5.6
  * SearchBar 1.6.3 (Disabled)
  * sheetmetal 0.7.54
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-09-20:
@PaddleStroke FYI
maybe related to https://github.com/FreeCAD/FreeCAD/issues/23886

depthoffocus, 2025-09-20:
Doesn't crash for me on macOS...

```
OS: macOS 15.7
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.75/2
Installed mods: 
  * ThreadProfile 1.96.0
  * freecad.gears 1.3.0
  * OpenTheme 2025.5.20
  * fasteners 0.5.41
  * lattice2 1.0.0
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.24
  * Curves 0.6.70
```

depthoffocus, 2025-09-20:
But it does crash on Kubuntu with an AppImage.

```
OS: Ubuntu 25.04 (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/134.063/1.5
```

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x458d0) [0x7bca814458d0]
#1  0x7bca838208a0 in SoSFFloat::setValue(float) from /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libCoin.so.80+0
#2  0x7bca87e2dcde in Gui::ViewProviderDatum::resetTemporarySize() from /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libFreeCADGui.so+0x7e
#3  0x7bc9dc6b8c49 in PartGui::TaskAttacher::~TaskAttacher() from /tmp/.mount_FreeCAeFdAde/usr/lib/PartGui.so+0x159
#4  0x7bc9dc6b8e0a in PartGui::TaskAttacher::~TaskAttacher() from /tmp/.mount_FreeCAeFdAde/usr/lib/PartGui.so+0xa
#5  0x7bca87d49f67 in Gui::TaskView::TaskDialog::~TaskDialog() from /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libFreeCADGui.so+0x37
#6  0x7bc9dcba25ca in PartDesignGui::TaskDlgPrimitiveParameters::~TaskDlgPrimitiveParameters() from /tmp/.mount_FreeCAeFdAde/usr/lib/PartDesignGui.so+0xa
#7  0x7bca87d44dca in Gui::TaskView::TaskView::removeDialog() from /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libFreeCADGui.so+0x16a
#8  0x7bca87d45b54 in Gui::TaskView::TaskView::reject() from /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libFreeCADGui.so+0x104
#9  /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libQt6Core.so.6(+0x31684a) [0x7bca81d1684a]
#10  /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libQt6Widgets.so.6(+0x3d8729) [0x7bca82fd8729]
#11  /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libQt6Core.so.6(+0x31684a) [0x7bca81d1684a]
#12  0x7bca83056727 in QAbstractButton::clicked(bool) from /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libQt6Widgets.so.6+0x47
#13  /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libQt6Widgets.so.6(+0x47be80) [0x7bca8307be80]
#14  /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libQt6Widgets.so.6(+0x4adba7) [0x7bca830adba7]
#15  0x7bca830add6d in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libQt6Widgets.so.6+0xbd
#16  0x7bca831c3864 in QWidget::event(QEvent*) from /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libQt6Widgets.so.6+0xa64
#17  0x7bca831b3e48 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libQt6Widgets.so.6+0x88
#18  0x7bca831b4cbd in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libQt6Widgets.so.6+0xdad
#19  0x7bca87953d78 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libFreeCADGui.so+0x78
#20  0x7bca81d5fb8a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libQt6Core.so.6+0x17a
#21  0x7bca831bbe7c in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libQt6Widgets.so.6+0x1ec
#22  /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libQt6Widgets.so.6(+0x5be8c7) [0x7bca831be8c7]
#23  /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libQt6Widgets.so.6(+0x5c5440) [0x7bca831c5440]
#24  0x7bca831b3e48 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libQt6Widgets.so.6+0x88
#25  0x7bca87953d78 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libFreeCADGui.so+0x78
#26  0x7bca81d5fb8a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libQt6Core.so.6+0x17a
#27  0x7bca823d1970 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libQt6Gui.so.6+0x810
#28  0x7bca82431e14 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libQt6Gui.so.6+0xb4
#29  /tmp/.mount_FreeCAeFdAde/usr/lib/qt6/plugins/platforms/../../../libQt6XcbQpa.so.6(+0x58f90) [0x7bca7dc6ef90]
#30  /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/./libglib-2.0.so.0(+0x5b646) [0x7bca7f2a6646]
#31  /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/./libglib-2.0.so.0(+0x5e817) [0x7bca7f2a9817]
#32  /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7bca7f2a9f2e]
#33  0x7bca81af6b12 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libQt6Core.so.6+0x72
#34  0x7bca81d9747c in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libQt6Core.so.6+0x25c
#35  0x7bca81d977da in QCoreApplication::exec() from /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libQt6Core.so.6+0x9a
#36  0x7bca8787c20d in Gui::Application::runApplication() from /tmp/.mount_FreeCAeFdAde/usr/bin/../lib/libFreeCADGui.so+0xc6d
#37  /tmp/.mount_FreeCAeFdAde/usr/bin/freecad(+0x9bb5) [0x5aa81197abb5]
#38  /lib/x86_64-linux-gnu/libc.so.6(+0x2a578) [0x7bca8142a578]
#39  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x7bca8142a63b]
#40  /tmp/.mount_FreeCAeFdAde/usr/bin/freecad(+0x8cd9) [0x5aa811979cd9]
```

chennes, 2025-09-21:
ASAN describes this as a "use-after-free", where `soScale` had been previously freed by 
```
2025-09-21 16:26:05.658996-0500 FreeCAD[24023:117269349] pc_0x10dce6348###func_Gui::ViewProviderPlane::~ViewProviderPlane()###file_ViewProviderPlane.cpp###line_62###obj_(libFreeCADGui.dylib:arm64+0x1dea348)
```

PaddleStroke, 2025-09-22:
@kadet1090 does that ring a bell? Maybe that comes from your recent changes to ViewProviderPlane?

closed 2025-09-25 by chennes