FreeCAD/FreeCAD #24091: <Import> ExportOCAF2.cpp(392): obj1#Variables has null shape

leoheck, 2025-09-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Exporting step complains of an object that has a null shape, but this object does not contain data.

```
14:50:36  <Import> ExportOCAF2.cpp(392): obj1#Variables has null shape
14:50:36  <Import> ExportOCAF2.cpp(392): obj2#Variables has null shape
14:50:36  <Import> ExportOCAF2.cpp(392): obj3#Variables has null shape
14:50:36  <Import> ExportOCAF2.cpp(392): obj4#Variables has null shape
14:50:36  <Import> ExportOCAF2.cpp(392): obj5#Variables has null shape
14:50:36  <Import> ExportOCAF2.cpp(392): obj6#Variables has null shape
14:50:36  <Import> ExportOCAF2.cpp(392): obj7#Variables has null shape
```

This is the type when asking on the console.

```
>>> obj.TypeId
'App::FeaturePython'
```

But the properties shows this:

[image]

The object is the Variables of the Asm4 workbench.

[image]

Is it possible make it stop complaining on things like this by default?

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
Installed mods: 
  * Assembly4.1 0.60.2.1
  * Cables 0.2.1
  * CurvesWB 0.6.64
  * FreeCAD-render 2024.12.15
  * sheetmetal 0.7.52
  * DynamicData 2.75.0
  * Curves 0.6.70
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2025.5.20
  * fasteners 0.5.41
  * Silk 0.2.0
  * osh-autodoc-workbench 0.2.3
  * CurvedShapes 1.0.13
  * freecad-povray-render
  * OpticsWorkbench 1.2.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.7.6
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-09-22:
I'm really not a fan of the word 'complains', the code output will no doubt be useful when attempting to find an issue with a specific Step file sometime in the future so for me the question **should** be '_Can the Warning level of this output please be dropped to Log/Message level instead?_' so it's not lost completely but also not interfering in the user's workflow.

leoheck, 2025-09-22:
@Syres916 This doesn’t affect my workflow, but the messages appear to indicate something that isn’t relevant for that type of object. For example, Asm4.Variables don’t have a shape. These messages don’t provide much value for debugging when they also appear for Parts without a shape, alongside others for objects that are already known to lack one.

FEA-eng, 2025-11-22:
@leoheck If it's still relevant, please share any FreeCAD file in which the issue occurs.

leoheck, 2025-11-22:
Sure, here it is [sample.FCStd.zip](https://github.com/user-attachments/files/23690986/sample.FCStd.zip)

The terminal shows this when exporting part to step.
```
15:35:59  <Import> ExportOCAF2.cpp(403): master#Variables has null shape
```

FEA-eng, 2026-02-21:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20260217 (Git shallow)
Build date: 2026/02/17 21:26:14
Build type: Release
Branch: (HEAD detached at 3bad5dabd)
Hash: 3bad5dabde38f1a901fa1475e30f5caee7f9846a
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

jumpjack, 2026-02-25:
I have same issue with this ancient NASA file from '90s, which I can't modify;
http://win98.altervista.org/pathfinder/mpo_latest_lo.zip
(unpack and  import **mpo_latest_lo.wrl** )

Any suggestion?

leoheck, 2026-02-26:
This issue is about the unnecessary message when exporting an object to step. 

The object generating that message should not be exported since it doesn't contain any shape. So why should it print any message.

Now, in your case, if nothing is getting exported to step, or a different format, then you should open an issue to discuss that.

FEA-eng, 2026-05-28:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20260526 (Git shallow)
Build date: 2026/05/26 17:25:08
Build type: Release
Branch: (HEAD detached at 26ca5b4bd)
Hash: 26ca5b4bda657d359a783e9de99d7ae0a07fc513
Python 3.11.15, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
```