FreeCAD/FreeCAD #24103: BIM: BIM workflow relies too much on the Property Editor

furgo16, 2025-09-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

This is not always the case, but I've noticed this often when working with BIM objects: whenever you need to edit them, often essential properties are not covered by their corresponding task panel. The result is either a back and forth travel from the task panel to the Property Editor, or extended back and forth scrolling to modify related properties that are in different locations in the Property editor, simply because the Property editor's properties are sorted alphabetically.

This is an example of the first case: double-clicking to edit a wall: I'd expect the task panel to give the ability to quickly edit its most essential properties:

- Height
- Length
- Width
- Alignment

Yet they're are not there, and the Property editor needs to be accessed instead:

[image]

If others agree, this could be used as a meta-issue to discuss the topic and potentially track task panel dialogs that could use incremental improvements.

This was also one of the goals of https://github.com/FreeCAD/FreeCAD/pull/23826 for the specific case of Section Planes.

### Full version info

```shell
1.1.0-dev
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ErtugrulEge, 2025-09-21:
Also when you go to property editor , the essential properties you covered are at the bottom of the properties panel

[image]

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: CAD/Free Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/127.628/1

semhustej, 2025-09-21:
That would make sense - to have the most changed properties in the edit task panel. I only rarely use the edit option to edit BIM objects using the edit tool.

I think a good way to start would be to create a spreadsheet with all BIM objects and list of properties, which could be placed in the task panel. That could be a good conversation starter and seeing multiple tools side by side should enhance consistency across different tools.

I can create such a spreadsheet, but I don't know when I will get to it.