FreeCAD/FreeCAD #24115: `part.section` returns incomplete (and wrong) results

freckletonj, 2025-09-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Here's a simple shape (i've tried to simplify it further, but `section` works more frequently if its much simpler):

[image]

I iteratively try to find the `section` of a cylinder with these top faces, and draw ones that show only partial results.

Here's what I expect, and most of the time it works like this:

[image]

Here are some failure cases, happening about 1:100. Note, it's not just when cylinder-faces have overlapping edges, sometimes the cylinder is in the middle of face, completely contained, and we get a partial result.

[image]

Here's the file (renamed `.FCStd -> .py` for github upload):

[t25_incomplete_section.py](https://github.com/user-attachments/files/22453812/t25_incomplete_section.py)

Here's the code you can run in the console to find bad sections:

```python
import random
import math

def sample_circle(radius):
    theta = random.random() * 2 * math.pi
    r = radius * math.sqrt(random.random())
    x = r * math.cos(theta)
    y = r * math.sin(theta)
    return x, y

part_radius = 100
tool_radius = 6
tool_height = 100

faces = FreeCAD.ActiveDocument.getObject('Binder').Shape

def find_bad_sections(n_tries=100):
    n_found = 0
    for _ in range(n_tries):
        x,y = sample_circle(part_radius - tool_radius)
        cyl = Part.makeCylinder(
            tool_radius,
            tool_height,
            FreeCAD.Vector(x, y, -tool_height/2),
            FreeCAD.Vector(0, 0, 1)
        )
        section = faces.section(cyl)
        section.tessellate(0.01)

        min_len = 2 * 3.14159 * tool_radius * 0.95
        if section.Length < min_len:  # len smaller than cross section of cylinder
            n_found += 1
            Part.show(section)  # show partial sections

        # Part.show(section)  # show all sections

    if n_found == 0:
        print('No bad sections found')
    else:
        print(f'found {n_found} bad sections and drew them')

find_bad_sections(500)

```
 

My "is bad" code simply checks if the section has a length smaller than a cross section of the cylinder. In practice, I can't do this since the cylinder may be hanging over a face edge, so the true part-cyl section will actually be smaller than that.

Any ideas how to fix this, or at least identify failure cases so I can retry?

(
and PS here's another `section` bug I found on the journey:
https://github.com/FreeCAD/FreeCAD/issues/24111
and OCC reports a possibly related (?) issue
https://github.com/Open-Cascade-SAS/OCCT/issues/275
)

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/09/02 21:58:19
Build type: Release
Branch: main
Hash: 04e085cba3b43d1e09aedd4445824a6eb46e45b6
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/127.628/1
Installed mods: 
  * ZFace
  * CurvedShapes 1.0.13
  * Curves 0.6.68
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

freckletonj, 2025-09-21:
If I take the failure cases, rotate the cylinder 180 around Z, and try again, together they seem to catch the right results (failure in blue, re-intersections in red)

[image]

freckletonj, 2025-09-21:
If I approximate a cylinder with a polygonal prism, it's a bit slower, but I haven't seen a failure yet:

```python
import random
import math
import time

def sample_circle(outer_radius, inner_radius=5):
    # Sample angle uniformly
    theta = random.random() * 2 * math.pi

    # Sample radius with correct distribution for uniform area sampling
    # We need to account for the "hole" in the middle
    u = random.random()
    r_squared = inner_radius**2 + u * (outer_radius**2 - inner_radius**2)
    r = math.sqrt(r_squared)

    # Convert to Cartesian coordinates
    x = r * math.cos(theta)
    y = r * math.sin(theta)
    return x, y

part_radius = 100
tool_radius = 6
tool_height = 100

faces = FreeCAD.ActiveDocument.getObject('Binder').Shape

def mk_poly_cylinder(radius, height, segments, center=FreeCAD.Vector(0,0,0)):
    pts = []
    for i in range(segments):
        angle = 2 * math.pi * i / segments
        x = center.x + radius * math.cos(angle)
        y = center.y + radius * math.sin(angle)
        pts.append(FreeCAD.Vector(x, y, center.z))
    pts.append(pts[0])  # close the loop

    wire = Part.makePolygon(pts)
    face = Part.Face(wire)
    solid = face.extrude(FreeCAD.Vector(0,0,height))
    return solid

def mk_section(x, y, z_rot, cyl_type):
    '''
    cyl_type in {'normal', 'polygon'}
    '''

    center = FreeCAD.Vector(x, y, -tool_height/2)

    if cyl_type == 'normal':
        # Part.makeCylinder
        cyl = Part.makeCylinder(
            tool_radius,
            tool_height,
            center,
            FreeCAD.Vector(0, 0, 1)
        )
    elif cyl_type == 'polygon':
        # Polygonal cylinder
        N_SEGMENTS = 32
        cyl = mk_poly_cylinder(tool_radius, tool_height, segments=N_SEGMENTS, center=center)
    else:
        raise ValueError(f'bad cyl_type: {cyl_type} must be "normal" or "polygon"')

    if z_rot is not None:
        center = FreeCAD.Vector(x, y, 0)
        cyl.rotate(center, FreeCAD.Vector(0, 0, 1), z_rot)
    section = faces.section(cyl)
    section.tessellate(0.01)
    return section

def find_bad_sections(n_tries=100, cyl_types=['normal', 'polygon']):
    for cyl_type in cyl_types:

        start = time.time()
        n_found = 0

        for _ in range(n_tries):
            x, y = sample_circle(part_radius - tool_radius, tool_radius + 5)  # 5mm is inner hole

            section = mk_section(x, y, z_rot=None, cyl_type=cyl_type)

            min_len = 2 * 3.14159 * tool_radius * 0.95
            if section.Length < min_len:  # len smaller than cross section of cylinder
                n_found += 1

                obj = Part.show(section)  # show partial sections
                obj.ViewObject.LineColor = (0.0, 0.0, 1.0)

                section2 = mk_section(x, y, z_rot=180, cyl_type=cyl_type)
                obj = Part.show(section2)
                obj.ViewObject.LineColor = (1.0, 0.0, 0.0)

            # Part.show(section)  # show all sections

        end = time.time()
        print(f'{cyl_type} time    : {(end - start):.1f} sec')
        print(f'{cyl_type} time per: {(end - start) / n_tries:.3f} sec')

        if n_found == 0:
            print(f'{cyl_type}: No bad sections found')
        else:
            print(f'{cyl_type}: found {n_found} bad sections and drew them')

find_bad_sections(1000, ['normal', 'polygon'])
```

maxwxyz, 2025-09-22:
@pieterhijma is this an OCC bug?

pieterhijma, 2025-09-24:
Yes, most likely.

freckletonj, 2025-09-24:
@pieterhijma You may have seen this already too then, not sure if it's related: https://github.com/Open-Cascade-SAS/OCCT/issues/275

As far as impact, `section` is a core player in some CAM work I'm doing - helping find collision bounds of an endmill with a part: https://forum.freecad.org/viewtopic.php?p=848689#p848689

And I intend to use this heavily in more CAM ops. I have multiple fall backs to get `section` to work (rotate the cylinder 180, use `section`'s `approximation=True`, use a polygonal approximation of a cylinder), and eventually it seems to work, but I'd love if it were more bullet proof.

_feature request:_ I'd also love to parallelize multiple cylinder intersections,  but if I intersect with a list of cylinders (which it does allow), I get back a differently-shaped list of edges (and different sorting?) so I can't tie the `section` results back to the cylinder responsible for those results, so, as more a feature request, that'd be nice if it had an api like `faces.section([cyl1, cyl2, cyl3, ...])` and gave back `[ [cyl1_edge1, cyl1_edge2], [cyl2_edge1], [<cyl3_didnt_intersect_so_empty_list_or_None>]) ]`.

pieterhijma, 2025-09-29:
@freckletonj, thanks. I recorded that in the project. As for the feature requests, could you please file separate issues for those?

pieterhijma, 2025-11-06:
> I see this problem still "needs triage", is there something I can do to help it move along? I think I've provided a pretty robust repro.

Yes, just other priorities. This one is also important but there are many waiting that also crash, for example. It would be very helpful if you can create a minimal example. See [this topic](https://forum.freecad.org/viewtopic.php?p=839428#p839428).

freckletonj, 2025-11-06:
@pieterhijma The top comment ⏫ has an `FCStd` and a console script for generating all the failed geometries (just open the FCStd in FreeCAD, open the FreeCAD python console, and copy paste the script).

I mention that if it's much more minimal than that, it doesn't fail as often, I'm happy to get on a call or something if that helps, or were you meaning I should reproduce this directly within OCC?

or maybe I was just confusing, I renamed the FCStd to `.py` so github would upload it: https://github.com/user-attachments/files/22453812/t25_incomplete_section.py  . So <- that's the file, and the script is in the original comment above.

freckletonj, 2025-11-07:
Thank you! And sorry, I didn't know that's what you meant - it *is* deterministic (but hard to determine when itll fail). With my script, you can `Part.show(...)` all the intersections, and see ones that fail, and delete the others. If you save that file, you now have geometries that will fail when intersected again.