FreeCAD/FreeCAD #24117: Linked bodies are not exported to Collada (.dae) when the parent part is exported

alpha3141, 2025-09-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When exporting a part to Collada (`.dae`) linked bodies don't get exported. Only the bodies that are present in the current document and aren't linked will be part of the `.dae` file. 
+ In the provided example the linked body "box" isn't present in the `.dae` export.

[image]

[dae_export_incomplete.zip](https://github.com/user-attachments/files/22454856/dae_export_incomplete.zip)

##### Image of the complete `.stl` export:
[image]

##### Image of the incomplete  `.dae` export:

[image]

### Full version info

```shell
OS: Fedora Linux 42 (Workstation Edition) (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/09/11 03:30:26
Build type: Release
Branch: main
Hash: 770f769b1f234d0b50bfd65f057c33c151622cd3
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Assembly4 0.50.18
  * fasteners 0.5.38
  * 3DfindIT 1.2.0 (Disabled)
  * Assembly3 0.12.2
  * OpenTheme 2025.5.20
  * freecad.cross 1.0.0
  * freecad.robotcad 9.1.2
  * WebTools 1.0.0
  * sheetmetal 0.7.24
  * addFC
  * freecad.gears 1.3.0

Incomplete exports also occur in Safe Mode.
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FEA-eng, 2025-11-22:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20251119 (Git shallow)
Build date: 2025/11/19 23:28:17
Build type: Release
Branch: main
Hash: deab0abf8e61d69f2208b794f64ac56428467fef
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

Gaurang-2005, 2026-03-01:
I reproduced this issue on the latest FreeCAD `main` branch.

Using a simple Body + App::Link example, exporting to Collada (.dae) results in two objects in Blender’s collection, but only one has geometry. The linked body appears as an empty object.

This suggests the Collada exporter creates nodes for `App::Link` objects but does not resolve and export their referenced geometry.

I’d like to work on this issue by investigating how the Collada exporter handles document objects and extending it to resolve `App::Link` references before export. Please let me know if there are any design considerations I should follow.