FreeCAD/FreeCAD #24132: PD: Thickness fail when there are round corners.

leoheck, 2025-09-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Thickness fails when there are round corners.

I have some quick tests here: [thickness.FCStd.zip](https://github.com/user-attachments/files/22470163/thickness.FCStd.zip)

[image]

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
Installed mods: 
  * Assembly4.1 0.60.2-0.1
  * Cables 0.2.1
  * CurvesWB 0.6.64
  * FreeCAD-render 2024.12.15
  * Assembly4 0.60.2
  * sheetmetal 0.7.55
  * DynamicData 2.75.0
  * Curves 0.6.70
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2025.5.20
  * fasteners 0.5.42
  * Silk 0.2.0
  * osh-autodoc-workbench 0.2.3
  * CurvedShapes 1.0.13
  * freecad-povray-render
  * OpticsWorkbench 1.2.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.7.6
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

aslze, 2026-03-11:
I also just noticed **Thickness** errors when there are **Spheres** in **PartDesign** overlapping other shapes (**FreeCAD 1.1 RC3**). Imagine a flat piece with a "dome" on top. Adding thickness selecting the bottom face fails. But if the sphere is replaced with a cylinder, and this cylinder is added big fillets to make it look like a spherical dome, then it works. It works as long as the fillet leaves a little circular face on top. If the fillet is made larger so that that top face becomes a point, then it fails.

Here is a sample file:

[spheres-thickess.zip](https://github.com/user-attachments/files/25920240/spheres-thickess.zip)

It contains two bodies. The first has spherical domes. If you select the bottom face and try Thickness it fails.
The second body has cylindrical domes, one of which is filleted to make it look spherical. And the Thickness operation works (both with rounding or with intersections). If the fillet radius is made larger, then Thickness fails too.

(I thought this was another example of the same issue, but I may be wrong).

stefano2734, 2026-09-12:
Increase your occt version to highest available. 7.8.1 is quite old here.
7.9.3 is last in 7er series compatible with version 1.1.3 and 26.3 dev.
Also 8-series is supported by 26.3 dev.
8.0.1 is here actual. Coming 8.02 and 8.1 will solve also FreeCAD problems.