FreeCAD/FreeCAD #24136: Error on open. All ufuncs...

dniezby, 2025-09-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

[image]

So, when I load in FreeCad this is the error that comes up. It doesn't seem to cause any issues. But it comes up. 

### Full version info

```shell
OS: Windows 11 build 26120
Architecture: x86_64
Version: 1.1.0dev.43071 (Git)
Build date: 2025/08/25 18:08:30
Build type: Release
Branch: main
Hash: d85613b498881c47485a7d930807673203086756
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Turntable/Window center
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.414/1.25
Installed mods: 
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

dniezby, 2025-09-22:
A month ago? I literally installed like 4 days ago.  But good point on updating again. I'll do that.

tarman3, 2025-09-22:
> A month ago?

Build date: 2025/08/25