FreeCAD/FreeCAD #24146: CAM:  Tapping Operation doesn't generate correct g-code

sliptonic, 2025-09-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Tapping was added as a new operation for version 1.1.  
The functionality is not mature and has several problems, especially in light of the new tool system.

- The tap tool configuration has a parameter for "TPI"  (Turns per inch).  This is clearly not unitless and will not work for metric taps
- The Operation does not have properties or logic to differentiate between rigid tapping and floating head tapping.  Therefore the internal representation doesn't have enough information for the postprocessors to correctly emit gcode.
- Postprocessors haven't been implemented and tested for tapping operations.

Tapping is desired functionality but is still incomplete.  Recommend we move tapping to an 'experimental feature' pending the 1.1 release and then improve it substantially for 1.2.

### Full version info

```shell
Any
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Reqrefusion, 2025-09-22:
> [@Reqrefusion](https://github.com/Reqrefusion) It's probably best to remove tapping from the release notes for 1.1

It's best to say it was added as an experimental feature with the latest PR. I've created a PR for it in the documentation repository.

sliptonic, 2025-09-24:
Feedback from a Gibbs user...  

Gibbs has the notion of a 'peck tap'.   I assume this is a tapping cycle with a chip breaking or clearing aspect.
I haven't seen this before but chip breaking is something I do all the time when hand-tapping.

[image]

luvtofish, 2026-02-11:
I've tested tapping in both 1.1rc2 and 1.2dev and G84/G85 tapping is working quite well with the dynapath_4060_post on my CNC vertical mill. 

With the changes made to the tool system, modifications were made to linuxcnc and copied to the dynapath_4060 post to accommodate G84/G85 Tapping. This change automates setting of the proper feed rate based on the RPM and pitch of the tool. Anyone working on other posts should be able to copy the G84/G85 code block and modify for their needs. 

A future update I'd like to see in the G84/G85 Tapping Op is a drop down menu for tap selection. This could be done in a similar way to how it has been implemented in the "Thread Milling Op".
 
@luvtofish

luvtofish, 2026-02-12:
@Connor9220. Oh wait. I think I may be referring to the ThreadMill Op. I’ve been bouncing between both and probably confused where I seen it.

luvtofish, 2026-02-12:
> The framework for testing dynapath_4060_post.py is now in PR [#27525](https://github.com/FreeCAD/FreeCAD/pull/27525), alongside a fix for handling --inches --precision=2. There is no test for the thread tapping support. I hope the code in [#26915](https://github.com/FreeCAD/FreeCAD/pull/26915) can provide some inspiration for writing such test.

I’ll need to have a look at at #26915 to understand this better. I’m not sure I would ever set precision = 2 but there could a use case out there somewhere.

petterreinholdtsen, 2026-03-01:
With the fixes in #27960 on top of the 1.1 branch, thread tapping work with the Fanuc post processor.