FreeCAD/FreeCAD #24181: Importing IFC files incorrect with newer versions of IfcOpenShell

sunxiaoling, 2025-09-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The IFC file was exported from Tekla Structures 19.0.
IfcOpenShell version 0.7.0.xxx imports the model correctly, but versions above 0.7.0 (0.7.1 to 0.8.3) import the model incorrectly — specifically, they fail to handle solid clipping properly.
(please rename out1.ifc.txt -> out1.ifc)

[out1.ifc.txt](https://github.com/user-attachments/files/22505307/out1.ifc.txt)

[image]

[image]

How can this issue be resolved?

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.2.39319 (Git)
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.12.11, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Chinese/China (zh_CN)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/windowsvista
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

sunxiaoling, 2025-09-24:
Thank you semhustej and maxwxyz  for confirming and labeling this issue!
This bug is blocking our workflow importing structural models from Tekla. 
Please let me know how I can assist further!

Moult, 2025-09-30:
I see someone has tested it on Bonsai and says it works, so in that case potentially the differences might be that FreeCAD probably uses the opencascade breps where we use the tessellation.

Dion Moult

sunxiaoling, 2025-10-04:
Hi @yorikvanhavre @semhustej ,
Thanks for looking into this!
From what I can see in the IFC file(line471~490), the H-beam (or plate) is indeed trimmed at both ends using two inclined clipping planes—approximately 10° and 5°—via IfcBooleanClippingResult.

#461= IFCAXIS2PLACEMENT3D(#459,#296,#460);
#462= IFCLOCALPLACEMENT(#32,#461);
#463= IFCRECTANGLEPROFILEDEF(.AREA.,'PL14*200',#53,14.,200.);
#464= IFCCARTESIANPOINT((1221.6537590124,7.27595761418343E-012,-8.92540850130735E-013));
#465= IFCAXIS2PLACEMENT3D(#464,#56,#55);
#466= IFCEXTRUDEDAREASOLID(#463,#465,#9,1245.35763242444);
#467= IFCDIRECTION((-0.984530483672433,-0.175213374831162,0.));
#468= IFCCARTESIANPOINT((1196.34772658083,19.3243677010396,-3.07627487927675E-008));
#469= IFCDIRECTION((0.,0.,-1.));
#470= IFCAXIS2PLACEMENT3D(#468,#467,#469);
#471= IFCPLANE(#470);
#472= IFCHALFSPACESOLID(#471,.T.);
**#473= IFCBOOLEANCLIPPINGRESULT(.DIFFERENCE.,#466,#472);**
#474= IFCDIRECTION((0.996124126500186,0.0879586528105232,0.));
#475= IFCCARTESIANPOINT((-3.77468722899721,21.9449362092564,4.52543645224068E-007));
#476= IFCAXIS2PLACEMENT3D(#475,#474,#9);
#477= IFCPLANE(#476);
#478= IFCHALFSPACESOLID(#477,.T.);
**#479= IFCBOOLEANCLIPPINGRESULT(.DIFFERENCE.,#473,#478);**
#480= IFCSHAPEREPRESENTATION(#12,'Body','Clipping',(#479));

I wonder if the recent updates in IfcOpenShell might have changed how IfcBooleanClippingResult is handled (or perhaps how it’s exposed to FreeCAD’s Native IFC importer). It seems the geometry is correctly defined in the IFC, but the clipping isn’t being applied during import, resulting in the full extrusion without the angled cuts.

Thanks again for your help!

yorikvanhavre, 2025-10-04:
Can someone try to use IfcConvert to convert the IFC file to STEP? That uses OCCT, but without FreeCAD. It should indicate us if it's a bug in IOS or in FreeCAD.

sunxiaoling, 2025-10-05:
The STEP model exported by IfcConvert 0.8.3 is incorrect, while the one exported by IfcConvert 0.7.11 is correct.

```
E:\Code>IfcConvert083 --convert-back-units out.ifc out.stp
IfcOpenShell IfcConvert 0.8.0 (OCC 7.8.1)
Scanning file...
Done scanning file
Done resolving references
Parsing input file took 0 seconds
Creating geometry...
Done creating geometry (17 objects)

Log:
[Error] [2025-10-05 16:27:40] {1bC8Hu000R9Z4sEJSsE34t} Failed to convert:
#149=IfcMaterial('STEEL/Q355C')
[Error] [2025-10-05 16:27:40] {1bC8Hu000R$Z4sEJSsE34t} Failed to convert:
#149=IfcMaterial('STEEL/Q355C')
[Error] [2025-10-05 16:27:40] {1bC8Hu000S0J4sEJSsE34t} Failed to convert:
#149=IfcMaterial('STEEL/Q355C')
[Error] [2025-10-05 16:27:40] {1bC8Hu000S134sEJSsE34t} Failed to convert:
#149=IfcMaterial('STEEL/Q355C')
[Error] [2025-10-05 16:27:40] {1bC8Hu000S1p4sEJSsE34t} Failed to convert:
#149=IfcMaterial('STEEL/Q355C')
[Error] [2025-10-05 16:27:40] {1bC8Hu000STZ4sEJSsE34t} Failed to convert:
#149=IfcMaterial('STEEL/Q355C')
[Error] [2025-10-05 16:27:40] {1bC8Hu000SY34sEJSsE34t} Failed to convert:
#149=IfcMaterial('STEEL/Q355C')
[Error] [2025-10-05 16:27:40] {1bC8Hu000SYp4sEJSsE34t} Failed to convert:
#149=IfcMaterial('STEEL/Q355C')
[Error] [2025-10-05 16:27:40] {1bC8Hu000SvJ4sEJSsE34t} Failed to convert:
#149=IfcMaterial('STEEL/Q355C')
[Error] [2025-10-05 16:27:40] {1bC8Hu000Sw34sEJSsE34t} Failed to convert:
#149=IfcMaterial('STEEL/Q355C')
[Error] [2025-10-05 16:27:40] {1bC8Hu000Swp4sEJSsE34t} Failed to convert:
#149=IfcMaterial('STEEL/Q355C')
[Error] [2025-10-05 16:27:40] {1bCTN$000MmJ4sEJStDp4v} Failed to convert:
#149=IfcMaterial('STEEL/Q355C')
[Error] [2025-10-05 16:27:40] {1bCTN$000Mn34sEJStDp4v} Failed to convert:
#149=IfcMaterial('STEEL/Q355C')
[Error] [2025-10-05 16:27:40] {1bCTN$000Mnp4sEJStDp4v} Failed to convert:
#149=IfcMaterial('STEEL/Q355C')
[Error] [2025-10-05 16:27:40] {1bD9Cb000$h34sEJSvDpOn} Failed to convert:
#149=IfcMaterial('STEEL/Q355C')
[Error] [2025-10-05 16:27:40] {1bFmD40001yp4sEJWsCp8s} Failed to convert:
#149=IfcMaterial('STEEL/Q355C')
[Error] [2025-10-05 16:27:40] {1bFmD40002SZ4sEJWsD3Kt} Failed to convert:
#149=IfcMaterial('STEEL/Q355C')

Conversion took 0 seconds

E:\Code>IfcConvert711 --convert-back-units out.ifc out.stp
IfcOpenShell IfcConvert 0.8.0 (OCC 7.7.1)
Scanning file...
Done scanning file
Parsing input file took 0 seconds
Creating geometry...
Done creating geometry (17 objects)

Conversion took 0 seconds
```
[out0711.stp.txt](https://github.com/user-attachments/files/22706995/out0711.stp.txt)
[out083.stp.txt](https://github.com/user-attachments/files/22706996/out083.stp.txt)
[image]
[image]

Roy-043, 2025-10-25:
> From what I can see in the IFC file(line471~490)

The quoted IFC code is from a different file.

furgo16, 2025-11-22:
Excellent! Coincidentally, I just upgraded the development snap package to `ifcopenshell` `0.8.4` to be able to build `aarch64` packages: https://github.com/FreeCAD/FreeCAD-snap/pull/258