FreeCAD/FreeCAD #24182: CAM: File always requires Recomputation

Connor9220, 2025-09-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In some cases, these attached files always want to recompute.  Looks like this only happens when I have Toolbits in the file.

I think this is a topo naming issue.

[image]

Attached are two files causing this issues.

[tap-test.FCStd.zip](https://github.com/user-attachments/files/22505560/tap-test.FCStd.zip)

[class-tray-1.1-v2.FCStd.zip](https://github.com/user-attachments/files/22505563/class-tray-1.1-v2.FCStd.zip)

IF these two files do not cause the error, then you may also need my toolbit shape files, which i can provide also.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.1.0dev.43556 +53 (Git)
Build date: 2025/09/23 17:59:51
Build type: Unknown
Branch: FixTapOperation,FixTapToolBit
Hash: 455932fc5c3fc6a073f072b9f8c46f7c4d9ab710
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Turntable/Object center
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
Installed mods: 
  * lattice2 1.0.0
  * cadenas3dfindit
  * ThreadProfile 1.96.0
  * A2plus 0.4.68
  * Ondsel-Lens 2024.11.29.01 (Disabled)
  * OpenTheme 2025.5.20
  * fasteners 0.5.41
  * dodo 1.0.1
  * QuickMeasure 2022.10.28
  * KM-FreeCAD 2025.7.27
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-09-25:
@drwho495 is this a TNP issue?

freckletonj, 2025-09-26:
i frequently see this unexpectedly too 👍

closed 2025-11-14 by Connor9220