FreeCAD/FreeCAD #24189: CAM Profile OP generates more path than selected bodies

leho-l, 2025-09-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When selecting a Job model body from tree view it should make profile operation for that
body only. at the moment it profiles all the bodies in the Job. Makes using helper bodies harder than it should.

[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.43611 (Git) Snap 1742
Build date: 2025/09/23 05:03:49
Build type: Release
Branch: main
Hash: c3707cc4a519a2b14d6dde3dadde525b3ba5350f
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.2, OCC 7.7.1
Locale: Estonian/Estonia (et_EE)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leho-l, 2025-09-24:
A workaround is selecting only the faces on the body, not the whole body.

leho-l, 2025-09-24:
[CAM multi body test.zip](https://github.com/user-attachments/files/22510474/CAM.multi.body.test.zip)

Made a simple example.
With this file its not able to make pocket operation, giving these errors:

[image]

tarman3, 2025-09-24:
Please check #24194
This changes will be added to next build [CAM3](https://github.com/tarman3/FreeCAD/releases), probably this weekend