FreeCAD/FreeCAD #24191: DRAFT: Slow, Slow, Slow

Luciniki, 2025-09-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Good morning everyone,
I've been working on this drawing imported from a DXF for about a month, and it's getting harder and harder to work with. Every little change takes an absurd amount of time.
Can anyone tell me if it's a DRAFT error or something wrong with the drawing?

remove .PDF from the attached file

[NIC.FCStd.PDF](https://github.com/user-attachments/files/22510432/NIC.FCStd.PDF)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.6656/1
Installed mods: 
  * A2plus 0.4.68
  * sheetmetal 0.7.24
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-09-24:
The number of document objects (almost 1000) is probably the bottleneck.

kaiwas, 2025-09-24:
> The number of document objects (almost 1000) is probably the bottleneck.

This only works for freecad. 
I work in Corel Draw and often combine multiple drawings on a single sheet.
For example, there are 30,000 objects in a document, yet I work without any significant delays.
Yes. This isn't a CAD system, but the example shown uses regular lines that don't require extensive calculations.

[image]

Perhaps the main reason is the single-threaded nature of the computations. With 4 physical cores (2x in hyperthreading mode), many tasks only load the computer by 1/8 (12%).

Luciniki, 2025-09-24:
By deleting the pages, it was actually accessible and editable again. Thanks.

Lgt2x, 2025-12-02:
Hello, can you please give this branch a try? https://github.com/FreeCAD/FreeCAD/pull/25898 It should solve the underlying issue

semhustej, 2025-12-03:
> I believe buliding FreeCAD is not trivial and some users wait for weekly releases to try things out. 

This is correct. I am also able to test new features only after they are merged.

Lgt2x, 2025-12-03:
The fix has not been merged yet btw, it's not included in the latest weekly build. Maybe the next one if it manages to go through review.

Luciniki, 2025-12-03:
> > Yes, I can try it as soon as the Development Build comes out weekly-2025.12.03
> 
> [@Luciniki](https://github.com/Luciniki) not sure I follow that logic, help me understand. How does waiting for the dev build have anything to do with testing Lgt2x's branch that requires compiling FreeCAD from source and inserting their fix to test?

Forgive me, but I don't have the skills to build it myself.
Maybe one day I'll find the time to learn how to do it :)