FreeCAD/FreeCAD #24206: Sketch Angular Deflection property has no effect

Beep6581, 2025-09-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Sketches have a property "Angular Deflection" in the "View" tab. It has no effect, so why have the property?

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.43651 (Git)
Build date: 2025/09/24 15:59:41
Build type: Release
Branch: main
Hash: 5194dfc18adcdcb01b1837c65cae6f005a54f8c0
Python 3.13.7, Qt 6.9.1, Coin 4.0.3, Vtk 9.5.1, boost 1_88, Eigen3 3.4.0, PySide 6.9.1
shiboken 6.9.1, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.9.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/108.888/1.5
Installed mods: 
  * OpenDark 2023.12.17
  * A2plus 0.4.68
  * fasteners 0.5.41
  * AddonManager 2025.8.27
  * CurvedShapes 1.0.13
  * woodworking 1.1.0.dev.1.1.1.2025.09.10
  * Curves 0.6.70
```

### Subproject(s) affected?

None

### Anything else?

https://github.com/user-attachments/assets/95a6249a-9a6e-45bd-b04c-522c9baf9e47

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-09-25:
Angular Deflection requires a `Recompute` of the sketch to take effect.

FEA-eng, 2026-01-06:
For me, it seems to take effect even before the input field loses focus:

[image]

luzpaz, 2026-04-08:
Needs to be retested on 1.1