FreeCAD/FreeCAD #24209: build fails with newer versions of pcl ie. pcl version 1.15.1

ipatch, 2025-09-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

i've working on building a freecad macos app bundle on macos ventura for the homebrew freecad tap. recently homebrew updated the pcl library / package to version 1.15.1 and since this release a header file has been renamed. so the offending file below will cause a build error.

```
src/Mod/ReverseEngineering/App/SurfaceTriangulation.cpp
```

the error message from the compiler,

```
/opt/code/git/github/forks/freecad-git/freecad-src/src/Mod/ReverseEngineering/App/SurfaceTriangulation.cpp:43:10: fatal error: 'pcl/point_traits.h' file not found
#include <pcl/point_traits.h>
         ^~~~~~~~~~~~~~~~~~~~
1 warning and 1 error generated.
make[2]: *** [src/Mod/ReverseEngineering/App/CMakeFiles/ReverseEngineering.dir/SurfaceTriangulation.cpp.o] Error 1
make[1]: *** [src/Mod/ReverseEngineering/App/CMakeFiles/ReverseEngineering.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

the easy fix is a guard for the new pcl header name, and build bumps along. ie.

```c
#if __has_include(<pcl/point_struct_traits.h>)
#include <pcl/point_struct_traits.h>
#else
#include <pcl/point_traits.h>
#endif
```

the above guard will keep the homebrew builds working while allowing other distributions that still use older versions of pcl to still build freecad.

i'm building freecad from a git clone and building the git tag v1.0.2

### Full version info

```shell
NA
```

### Subproject(s) affected?

None

### Anything else?

NA

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-09-25:
does the latest main work?

Syres916, 2025-09-25:
@ipatch see the change made to `main` in March (IMHO it should have been backported) https://github.com/FreeCAD/FreeCAD/commit/d9e731ca94abc14808ebeed208617116f6d5ea4a

ipatch, 2025-09-25:
@Syres916 so you're saying the pcl library replaced the `point_traits.h` header with `type_traits.h` ?

and for what it is worth i successfully built freecad using the header i mentioned above, ie. `point_struct_traits.h`

but it looks the header you mentioned ie. the header used the current source of freecad ie. `type_traits.h` does import / include the header i mentioned above,

`type_traits.h`

```c
#pragma once

#include <pcl/point_struct_traits.h>  // for pcl::traits::POD, pcl::traits::name, pcl::traits::datatype, pcl::traits::offset

#include <cstddef>  // for std::size_t
#include <cstdint>  // for std::uint8_t
```

and looking the pcl github source repo, it looks like the offending header was removed in version 1.13 and the last version that had it was in version 1.12

https://github.com/PointCloudLibrary/pcl/tree/pcl-1.13.0/common/include/pcl

https://github.com/PointCloudLibrary/pcl/tree/pcl-1.12.1/common/include/pcl