FreeCAD/FreeCAD #24210: PartDesing: Some properties are not editable anymore.

leoheck, 2025-09-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Some properties are not editable with Side 1 being `through all`.

For instance, I should be able to edit `Offset` and `Taper angle`.

This is maybe a regression after the refactor that happened with their symmetrical stuff. 

Someone should review these things for at least Pat/Pocket. 

[image]

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
Installed mods: 
  * Assembly4.1 0.60.2-0.1
  * Cables 0.2.1
  * CurvesWB 0.6.64
  * FreeCAD-render 2024.12.15
  * Assembly4 0.60.2
  * sheetmetal 0.7.55
  * DynamicData 2.75.0
  * Curves 0.6.70
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * Assembly4.1_mod 0.60.2-0.1
  * OpenTheme 2025.5.20
  * fasteners 0.5.42
  * Silk 0.2.0
  * osh-autodoc-workbench 0.2.3
  * CurvedShapes 1.0.13
  * freecad-povray-render
  * OpticsWorkbench 1.2.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.7.6
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-09-27:
@PaddleStroke is this by accident?

PaddleStroke, 2025-09-29:
Yes I'll check this out

PaddleStroke, 2025-09-29:
[image]

In 1.0 in ThroughAll the Offset spinbox is visible but disabled. I don't understand what it is supposed to do.
In 1.0 in the property view the property does not look like it is disabled, but if you click on it then it cannot be set.

What am I missing about offset? What is it supposed to do? Why I can't seem to be able to use it even in 1.0 ?

About Taper angle yes it should be enabled.

PaddleStroke, 2025-09-29:
I'm having a look at the code before my '2 sides' PR and I see that : 
Offset was already disabled before my PR.

If I understand correctly the code, offset was offseting the start of the pocket before it went through all. Can you confirm if I got this right?

If yes it seems that we do not need this anymore. Since now the normal way to do this would be to use 2 sides, put through all in one direction and length to second side with -offset value.

So my question is, should I remove offset from ThroughAll. Or should I restore the functionality that was already broken in 1.0?

PaddleStroke, 2025-09-29:
[image]

See in this picture, the offset functionality can be done with the 2 sides now. Imo offset should be removed from through all. No point keeping several ways to achieve the same.

PaddleStroke, 2025-09-30:
Actually offset has been disabled and non-working for throughAll for a long time apparantly. So I don't think a migration script is necessary.