FreeCAD/FreeCAD #24212: DXF exported drawing from tech draw dimensions wrong

Wallabygeorge, 2025-09-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I am using Freecad 1.0.2 and last week the drawings that were exported as a .DXF were correct when opened with libre cad. This week they are smaller. A 20 mm dimension shows up as 3 mm, which is a factor of 0.15. 

I am running linux mint version 22.1

Unfortunately Freecad files and dxf files are not supported. 

Help. 

Thanks

### Full version info

```shell
OS: KDE Flatpak runtime (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.2.39319 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 256fc7e)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.12.10, Qt 6.9.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Installed mods: 
  * A2plus 0.4.68
  * sheetmetal 0.7.24
  * fasteners 0.5.41
  * ExplodedAssembly
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Wallabygeorge, 2025-09-25:
An additional note The exported files exported before the problem occurred open in libre cad with correct dimensions. I believe that libre cad is not the problem.

maxwxyz, 2025-09-25:
can you reproduce the issue in the latest weekly: https://github.com/FreeCAD/FreeCAD/releases

closed 2025-09-26 by maxwxyz