FreeCAD/FreeCAD #24219: UI: The custom theme pack isn't fully applied.

kaiwas, 2025-09-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

How to reproduce:
1) Change the 3D background color in the dark theme (for clarity in the next steps).
2) Save the theme under any name.
3) Change the interface to a light or classic theme (let's say you're tired of the dark theme).
4) You're tired of the classic theme and want to apply your custom theme pack. Click Apply PACK. You'll see that your saved theme hasn't been fully applied.
5) Restarting won't help.

The light or classic theme will be displayed in the settings.

https://github.com/user-attachments/assets/3214cda2-b656-4568-9d83-bb65b10f8039

One workaround: Select the dark theme from the drop-down list and then apply the custom theme pack.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * addFC-main 1.6.2
  * Reinforcement
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * LyapinWorkbrench
  * FreeCAD_FastenersWB-master 0.5.42
  * FreeCAD_SketchArch-master
  * FrameForge 0.1.3
  * Road 2025.2.22
  * Curves 0.6.70
  * freecad.gears 1.3.0
  * CurvedShapesWorkbench-master 1.0.13
  * sheetmetal 0.7.53
  * OpenTheme 2025.5.20
  * FreeCAD_ThreadProfile-master 1.97.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

obelisk79, 2025-10-06:
I am unable to recreate this behavior. I change the background color, save the preference pack, switch themes, apply the saved preference pack, and background color is restored to what I saved.

kaiwas, 2025-10-06:
> Hey [@kaiwas](https://github.com/kaiwas) are you on Discord?

In my country, they're blocking it for violating some laws. So Discord works with difficulty and not always.
This afternoon, I managed to write a couple of messages in the DWG thread about FreeCAD. But now it's down again.