FreeCAD/FreeCAD #24244: CAM Array: issues with multiple ops in one array

freckletonj, 2025-09-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Based on discord [chat](https://discord.com/channels/870877411049357352/1162574657694728254/1420928243451691049) I'm adding a few issues with the CAM Array:

1. Array supports multiple ops but this isn't documented anywhere
2. the order that the array performs the ops depends on the order you selected them. If this is a feature, it must be documented, and editing the underlying features must make it clear what order they'll be performed in, and allow you to reorder. This hit me in an actual job on the CNC, where it reordered the ops that were performed
3. if I shift+select multiple ops to place in the Array, the base path of dressups get added. This separately bit me on the CNC, where both the dressup and its base were performed, destroying a part.
4. editing the base ops in an array sometimes doesn't actually apply the new paths, but in my attached repro, they *did* work, so I'm struggling to reproduce this specific failure.

Here's a repro, renamed `.FCStd -> .py` for github upload to work: 

[t26_array_issues.py](https://github.com/user-attachments/files/22550185/t26_array_issues.py)

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/09/02 21:58:19
Build type: Release
Branch: main
Hash: 04e085cba3b43d1e09aedd4445824a6eb46e45b6
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/127.628/1
Installed mods: 
  * ZFace
  * CurvedShapes 1.0.13
  * Curves 0.6.68
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tarman3, 2025-09-26:
2. Order depend from selection
This is standard behavior of all operation

There is no Task panel, so reorder can be done only by one way:
clear selection in property view and select operations in new order

3. For me this behavior is absolutely logical and correct
Operation and dressup above it is two different operation
If you select both, you get both in array
Probably you should select operation more precisely by CTRL

freckletonj, 2025-09-26:
i admit I see the logic of it too after the fact, thinking from the code pov, but only after it bit me in 2 different ways on the actual production runs. and it was hard to see these ahead of time either inspecting the g-code (of a multi MB file), or in the simulator (since the Array ops happen visually away from the stock, which I could have increased the stock size, but that would have messed up some of the ops for a single part run which is stock aware)

and since it'd be nice to not get bit by it for those who are unaware, I'd propose:

1. documentation that this is supported, and how it works
2. in the edit pop up, include numbers indicating the order of the ops, or a task panel that shows what's going on

I think that'd cover it.

tarman3, 2025-09-26:
I agree

tarman3, 2025-09-26:
As `Array` tool do have Task panel you can see order of selection and processing in `Property view`

[image]