FreeCAD/FreeCAD #24247: PartDesign: Datum plane failure and crash when cancelling datum creation between two faces of a short cylinder

FEA-eng, 2025-09-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

1. Create default Additive Cylinder.
2. Select its top and bottom face.
3. Create datum plane. It will fail with:
`09:24:26  PositionBySupport: AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor is trivial, principal axes are undefined.`
4. Now click Cancel. FreeCAD will crash (on both Ubuntu and Windows)

It also happens with manually created (padded) cylinder of the same size. But doesn't happen when the height is changed to 20 mm - then datum plane is correctly placed in the middle.

Datum plane errors also occur in 1.0.2 but without crashing.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-09-26:
Crash confirmed with IMHO no useful trace:

```
PositionBySupport: AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor is trivial, principal axes are undefined.
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45330) [0x7ed466645330]
#1  [0x610337de6e00]
```

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.43680 (Git)
Build date: 2025/09/25 12:37:38
Build type: Release
Branch: main
Hash: 864cde5aedab02d157853fd63865cca02049ed6d
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.13
shiboken 5.15.13, SMESH 7.7.1.0, xerces-c 3.2.4, IfcOpenShell 0.8.2, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: Behave-dark.qss/FreeCAD Classic/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/95.9567/1
Installed mods: 
  * FreeCAD-themes 2025.6.4
  * FreeCAD-Reporting
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * CfdOF 1.30.4
  * Curves 0.6.64
  * sheetmetal 0.7.23
```

xtemp09, 2025-09-26:
Here is more meaningful backtrace.

<details>
<summary>Backtrace</summary>

```cpp
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=<optimized out>, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007ffff1c4527e in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007ffff1c288ff in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007ffff1c2881b in __assert_fail_base (
    fmt=0x7ffff1dd01e8 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", 
    assertion=assertion@entry=0x7ffff64288f8 "doc != d->documents.end()", 
    file=file@entry=0x7ffff6385e60 "/home/xtemp09/Downloads/FreeCAD/freecad-source/src/Gui/Application.cpp", 
    line=line@entry=1067, 
    function=function@entry=0x7ffff63864e8 "void Gui::Application::slotShowHidden(const App::Document&)")
    at ./assert/assert.c:96
#6  0x00007ffff1c3b517 in __assert_fail (assertion=0x7ffff64288f8 "doc != d->documents.end()", 
    file=0x7ffff6385e60 "/home/xtemp09/Downloads/FreeCAD/freecad-source/src/Gui/Application.cpp", line=1067, 
    function=0x7ffff63864e8 "void Gui::Application::slotShowHidden(const App::Document&)")
    at ./assert/assert.c:105
#7  0x00007ffff5b2972c in Gui::Application::slotShowHidden (this=0x7fffffffd350, Doc=...)
    at /home/xtemp09/Downloads/FreeCAD/freecad-source/src/Gui/Application.cpp:1067
#8  0x00007ffff5b57fab in std::__invoke_impl<void, void (Gui::Application::*&)(App::Document const&), Gui::Application*&, App::Document const&> (__f=<optimized out>, __t=<optimized out>)
    at /usr/include/c++/14/bits/invoke.h:74
#9  0x00007ffff5b57fc9 in std::__invoke<void (Gui::Application::*&)(App::Document const&), Gui::Application*&, App::Document const&> (__fn=<optimized out>) at /usr/include/c++/14/bits/invoke.h:90
#10 std::_Bind<void (Gui::Application::*(Gui::Application*, std::_Placeholder<1>))(App::Document const&)>::__call<void, App::Document const&, 0ul, 1ul>(std::tuple<App::Document const&>&&, std::_Index_tuple<0ul, 1ul>) (this=<optimized out>, __args=...) at /usr/include/c++/14/functional:513
#11 std::_Bind<void (Gui::Application::*(Gui::Application*, std::_Placeholder<1>))(App::Document const&)>::operator()<App::Document const&, void>(App::Document const&) (this=<optimized out>)
    at /usr/include/c++/14/functional:598
#12 boost::detail::function::void_function_obj_invoker1<std::_Bind<void (Gui::Application::*(Gui::Application*, std::_Placeholder<1>))(App::Document const&)>, void, App::Document const&>::invoke(boost::detail::function::function_buffer&, App::Document const&) (function_obj_ptr=..., a0=...)
    at /usr/include/boost/function/function_template.hpp:158
#13 0x00007ffff4f0cdad in boost::function1<void, bool>::operator() (this=<optimized out>, a0=<optimized out>)
    at /usr/include/boost/function/function_template.hpp:771
#14 0x00007ffff4f0ce6e in boost::signals2::detail::call_with_tuple_args<boost::signals2::detail::void_type>::m_invoke<boost::function<void (bool)>, 0u, bool&>(boost::function<void (bool)>&, boost::signals2::detail::unsigned_meta_array<0u>, std::tuple<bool&> const&, boost::enable_if<boost::is_void<boost::function<void (bool)>::result_type>, void>::type*) const (this=<optimized out>, func=..., args=std::tuple containing = {...})
    at /usr/include/boost/signals2/detail/variadic_slot_invoker.hpp:105
#15 boost::signals2::detail::call_with_tuple_args<boost::signals2::detail::void_type>::operator()<boost::function<void (bool)>, bool&, 1ul>(boost::function<void (bool)>&, std::tuple<bool&> const&, mpl_::size_t<1ul>) const (this=<optimized out>, func=..., args=std::tuple containing = {...})
    at /usr/include/boost/signals2/detail/variadic_slot_invoker.hpp:90
#16 boost::signals2::detail::variadic_slot_invoker<boost::signals2::detail::void_type, bool>::operator()<boost::shared_ptr<boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot<void (bool), boost::function<void (bool)> >, boost::signals2::mutex> > >(boost::shared_ptr<boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot<void (bool), boost::function<void (bool)> >, boost::signals2::mutex> > const&) const (this=this@entry=0x7fffffffb240, connectionBody=...)
    at /usr/include/boost/signals2/detail/variadic_slot_invoker.hpp:133
#17 0x00007ffff4f0d3e7 in boost::signals2::detail::slot_call_iterator_t<boost::signals2::detail::variadic_slot_invoker<boost::signals2::detail::void_type, bool>, std::_List_iterator<boost::shared_ptr<boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot<void (bool), boost::function<void (bool)> >, boost::signals2::mutex> > >, boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot<void (bool), boost::function<void (bool)> >, boost::signals2::mutex> >::dereference() const (
    this=this@entry=0x7fffffffb020) at /usr/include/boost/signals2/detail/slot_call_iterator.hpp:110
#18 0x00007ffff4f0e0ce in boost::iterators::iterator_core_access::dereference<boost::signals2::detail::slot_call_iterator_t<boost::signals2::detail::variadic_slot_invoker<boost::signals2::detail::void_type, bool>, std::_ [... cut ...]
Build date: 2025/09/25 13:46:41
Build type: Unknown
Branch: master
Hash: 702eb9f5af0024080520fcd80cac8a6fbf682bc1
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.13
shiboken 5.15.13, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.6.3
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96.0584/96.0584/1
```

</details>

depthoffocus, 2025-09-26:
@kadet1090 @chennes without a stack trace it's not possible to be sure, but a crash when cancelling is reminiscent of #24077 and #23886

xtemp09, 2025-09-26:
@depthoffocus, 👆

depthoffocus, 2025-09-26:
I have actually managed to get a crash from this on the Mac now -- it is intermittent though. Maybe my initial hunch above was right.

```
Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libCoin.80.0.3.dylib          	       0x10d7e7ce4 SoSFFloat::setValue(float) + 4
1   libFreeCADGui.dylib           	       0x109ded419 Gui::ViewProviderDatum::resetTemporarySize() + 121
2   PartGui.so                    	       0x1331a668b PartGui::TaskAttacher::~TaskAttacher() + 331
3   PartGui.so                    	       0x1331a686e PartGui::TaskAttacher::~TaskAttacher() + 14
4   libFreeCADGui.dylib           	       0x109cf0627 Gui::TaskView::TaskDialog::~TaskDialog() + 71
5   PartGui.so                    	       0x1331ac401 PartGui::TaskDlgAttacher::~TaskDlgAttacher() + 161
6   libFreeCADGui.dylib           	       0x109ceba94 Gui::TaskView::TaskView::removeDialog() + 340
7   libFreeCADGui.dylib           	       0x109cec624 Gui::TaskView::TaskView::reject() + 244
8   libQt6Core.6.8.3.dylib        	       0x109117beb void doActivate<false>(QObject*, int, void**) + 1579
9   libQt6Widgets.6.8.3.dylib     	       0x10c03ecc6 QDialogButtonBoxPrivate::handleButtonClicked() + 198
10  libQt6Core.6.8.3.dylib        	       0x109117beb void doActivate<false>(QObject*, int, void**) + 1579
11  libQt6Widgets.6.8.3.dylib     	       0x10c0d47a3 QAbstractButtonPrivate::emitClicked() + 115
12  libQt6Widgets.6.8.3.dylib     	       0x10c0d44e7 QAbstractButtonPrivate::click() + 247
13  libQt6Widgets.6.8.3.dylib     	       0x10c0d60e1 QAbstractButton::mouseReleaseEvent(QMouseEvent*) + 209
14  libQt6Widgets.6.8.3.dylib     	       0x10c15b551 QWidget::event(QEvent*) + 2401
15  libQt6Widgets.6.8.3.dylib     	       0x10c220948 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 312
16  libQt6Widgets.6.8.3.dylib     	       0x10c222e7e QApplication::notify(QObject*, QEvent*) + 6014
17  libFreeCADGui.dylib           	       0x10975ba1c Gui::GUIApplication::notify(QObject*, QEvent*) + 92
18  libQt6Core.6.8.3.dylib        	       0x1091cd602 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 162
19  libQt6Widgets.6.8.3.dylib     	       0x10c220f5b QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) + 875
20  libQt6Widgets.6.8.3.dylib     	       0x10c18d988 QWidgetWindow::handleMouseEvent(QMouseEvent*) + 2040
21  libQt6Widgets.6.8.3.dylib     	       0x10c18c613 QWidgetWindow::event(QEvent*) + 115
22  libQt6Widgets.6.8.3.dylib     	       0x10c220948 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 312
23  libQt6Widgets.6.8.3.dylib     	       0x10c2218ee QApplication::notify(QObject*, QEvent*) + 494
24  libFreeCADGui.dylib           	       0x10975ba1c Gui::GUIApplication::notify(QObject*, QEvent*) + 92
25  libQt6Core.6.8.3.dylib        	       0x1091cd602 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 162
26  libQt6Gui.6.8.3.dylib         	       0x10ca28102 QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) + 2562
27  libQt6Gui.6.8.3.dylib         	       0x10c972e3b QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 459
28  libqcocoa.dylib               	       0x10ed3b6b0 QCocoaEventDispatcherPrivate::postedEventsSourceCallback(void*) + 448
29  CoreFoundation                	    0x7ff80502aab0 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
30  CoreFoundation                	    0x7ff80502aa52 __CFRunLoopDoSource0 + 157
31  CoreFoundation                	    0x7ff80502a80d __CFRunLoopDoSources0 + 203
32  CoreFoundation                	    0x7ff805029476 __CFRunLoopRun + 973
33  CoreFoundation                	    0x7ff805028a72 CFRunLoopRunSpecific + 536
34  HIToolbox                     	    0x7ff810b1a0d4 RunCurrentEventLoopInMode + 281
35  HIToolbox                     	    0x7ff810b1ce60 ReceiveNextEventCommon + 188
36  HIToolbox                     	    0x7ff810ca619a _BlockUntilNextEventMatchingListInModeWithFilter + 63
37  AppKit                        	    0x7ff808a8ddc1 _DPSNextEvent + 912
38  AppKit                        	    0x7ff80951c097 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1263
39  AppKit                        	    0x7ff808a7eea9 -[NSApplication run] + 610
40  libqcocoa.dylib               	       0x10ed38e28 QCocoaEventDispatcher::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 2808
41  libQt6Core.6.8.3.dylib        	       0x1091ce296 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 534
42  libQt6Core.6.8.3.dylib        	       0x1091cdefc QCoreApplication::exec() + 156
43  libFreeCADGui.dylib           	       0x1096708cf Gui::Application::runApplication() + 2415
44  freecad                       	       0x106e70b9e main + 3646
45  dyld                          	    0x7ff804b9b530 start + 3056
```

xtemp09, 2025-09-26:
I git pulled newer commits and now the stack trace looks like the one provided by depthoffocus above.

<details>
<summary>Backtrace
</summary>

```cpp
#0  0x0000000000000000 in ?? ()
#1  0x00007ffff159daac in SoField::valueChanged(int) () from /lib/x86_64-linux-gnu/libCoin.so.80c
#2  0x00007ffff618dca2 in SoSFFloat::operator= (this=<optimized out>, newvalue=1)
    at /usr/include/Inventor/fields/SoSFFloat.h:42
#3  Gui::ViewProviderDatum::resetTemporarySize (this=this@entry=0x55555ac27140)
    at /home/xtemp09/Downloads/FreeCAD/freecad-source/src/Gui/ViewProviderDatum.cpp:146
#4  0x00007fffa20782cb in PartGui::TaskAttacher::~TaskAttacher (this=0x55555917c480, 
    __in_chrg=<optimized out>)
    at /home/xtemp09/Downloads/FreeCAD/freecad-source/src/Mod/Part/Gui/TaskAttacher.cpp:271
#5  0x00007fffa20783ed in PartGui::TaskAttacher::~TaskAttacher (this=0x55555917c480, 
    __in_chrg=<optimized out>)
    at /home/xtemp09/Downloads/FreeCAD/freecad-source/src/Mod/Part/Gui/TaskAttacher.cpp:274
#6  0x00007ffff609f61a in Gui::TaskView::TaskDialog::~TaskDialog (this=this@entry=0x55555aceac60, 
    __in_chrg=<optimized out>)
    at /home/xtemp09/Downloads/FreeCAD/freecad-source/src/Gui/TaskView/TaskDialog.cpp:60
#7  0x00007fffa207127f in PartGui::TaskDlgAttacher::~TaskDlgAttacher (this=0x55555aceac60, 
    __in_chrg=<optimized out>)
    at /home/xtemp09/Downloads/FreeCAD/freecad-source/src/Mod/Part/Gui/TaskAttacher.cpp:1284
#8  0x00007fffa207129b in PartGui::TaskDlgAttacher::~TaskDlgAttacher (this=0x55555aceac60, 
    __in_chrg=<optimized out>)
    at /home/xtemp09/Downloads/FreeCAD/freecad-source/src/Mod/Part/Gui/TaskAttacher.cpp:1284
#9  0x00007ffff6099dc7 in Gui::TaskView::TaskView::removeDialog (this=this@entry=0x555555f18740)
    at /home/xtemp09/Downloads/FreeCAD/freecad-source/src/Gui/TaskView/TaskView.cpp:708
#10 0x00007ffff609a13c in Gui::TaskView::TaskView::reject (this=0x555555f18740)
    at /home/xtemp09/Downloads/FreeCAD/freecad-source/src/Gui/TaskView/TaskView.cpp:920
#11 0x00007ffff609bfc4 in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (Gui::TaskView::TaskView::*)()>::call(void (Gui::TaskView::TaskView::*)(), Gui::TaskView::TaskView*, void**) (f=<optimized out>, o=<optimized out>, arg=<optimized out>)
    at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:152
#12 0x00007ffff609c002 in QtPrivate::FunctionPointer<void (Gui::TaskView::TaskView::*)()>::call<QtPrivate::List<>, void>(void (Gui::TaskView::TaskView::*)(), Gui::TaskView::TaskView*, void**) (f=<optimized out>, 
    o=<optimized out>, arg=<optimized out>)
    at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:185
#13 QtPrivate::QSlotObject<void (Gui::TaskView::TaskView::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) (which=<optimized out>, this_=<optimized out>, 
    r=<optimized out>, a=<optimized out>, ret=<optimized out>)
    at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:418
#14 0x00007ffff2712e16 in ?? () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#15 0x00007ffff3518188 in ?? () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16 0x00007ffff2712dbf in ?? () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#17 0x00007ffff34676e4 in QAbstractButton::clicked(bool) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#18 0x00007ffff346796f in ?? () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#19 0x00007ffff34695b0 in ?? () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#20 0x00007ffff34697b8 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) ()
   from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#21 0x00007ffff33b0df8 in QWidget::event(QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#22 0x00007ffff336bd45 in QApplicationPrivate::notify_helper(QObject*, QEvent*) ()
   from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#23 0x00007ffff33746b0 in QApplication::notify(QObject*, QEvent*) ()
   from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#24 0x00007ffff5bfe1f3 in Gui::GUIApplication::notify (this=0x7fffffffd2e0, receiver=0x55555aeaee50, 
    event=0x7fffffffc590) at /home/xtemp09/Downloads/FreeCAD/freecad-source/src/Gui/GuiApplication.cpp:96
#25 0x00007ffff26d8118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) ()
   from /lib/x86_64-linux-gnu/libQt5Core.so.5
#26 0x00007ffff3372874 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#27 0x00007ffff33caa39 in ?? () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#28 0x00007ffff33cdfbf in ?? () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#29 0x00007ffff336bd45 in QApplicationPrivate::notify_helper(QObject*, QEvent*) ()
   from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#30 0x00007ffff5bfe1f3 in Gui::GUIApplication::notify (this=0x7fffffffd2e0, receiver=0x555555f07c30, 
    event=0x7fffffffca80) at /home/xtemp09/Downloads/FreeCAD/freecad-source/src/Gui/GuiApplication.cpp:96
#31 0x00007ffff26d8118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) ()
   from /lib/x86_64-linux-gnu/libQt5Core.so.5
#32 0x00007ffff2b45a3b in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () from /lib/x86_64-linux-gnu/libQt5Gui.so.5
#33 0x00007ffff2b17bfc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt5Gui.so.5
[... cut ...]
Version: 1.1.0dev.43681 (Git)
Build date: 2025/09/25 13:46:41
Build type: Unknown
Branch: master
Hash: 702eb9f5af0024080520fcd80cac8a6fbf682bc1
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.13
shiboken 5.15.13, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.6.3
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/unset/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96.0584/96.0584/1
```

</details>

kadet1090, 2025-09-28:
Yep, the backtrace seems to indicate that exact problem. I'll close it - feel free to repoen if it was not fixed.

closed 2025-09-28 by kadet1090