FreeCAD/FreeCAD #24253: UI/UX/Gui: Add a direction marker to sketch lines.

kaiwas, 2025-09-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Currently, there's no way to identify the start and end of a line segment.
While this was previously only important for BIM objects (like walls), the sketcher itself now has direction-sensitive tools (clockwise or counterclockwise).

In this video, I demonstrate how sketch direction affects the offset tool.

If the directions match, the offset is constructed without problems.
If the directions are the same, the offset is constructed without problems.

https://github.com/user-attachments/assets/ee365973-1052-46f9-8227-648ef789cdba

It's likely a long and arduous process to improve the offset algorithms. BUT! We could enable users to "fix" sketches themselves.

This requires two things:
1) Line direction markers.
This option may be optional and can be enabled in the element filter.
[image]

2) Add a direction change tool.
I described its operation in another issue.  https://github.com/FreeCAD/FreeCAD/issues/11957

All this will allow the user to analyze directions and change them independently WITHOUT rebuilding the entire geometry anew.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * addFC-main 1.6.2
  * Reinforcement
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * LyapinWorkbrench
  * FreeCAD_FastenersWB-master 0.5.42
  * FreeCAD_SketchArch-master
  * FrameForge 0.1.3
  * Road 2025.2.22
  * Curves 0.6.70
  * freecad.gears 1.3.0
  * CurvedShapesWorkbench-master 1.0.13
  * sheetmetal 0.7.53
  * OpenTheme 2025.5.20
  * FreeCAD_ThreadProfile-master 1.97.0
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leoheck, 2025-09-26:
Interesting. 

> All this will allow the user to analyze directions and change them independently WITHOUT rebuilding the entire geometry anew.

Depending on how discrete the mark is, this could always be enabled or enabled by default. 
This could give more visual info to devs when analysing issues.

leoheck, 2025-09-26:
Initially, it looks fine to have it this way. But then it is going to conflict with the mid line constraint `><`.
It could be an arrow pointing to the end, maybe.

[image]

[image]

kaiwas, 2025-09-26:
Yes, we need to think about the design. I just described the operating principle.

belinski-roman, 2025-09-26:
> Поначалу все выглядит хорошо. Но тогда это будет конфликтовать с ограничением средней линии 

You can create two labels per segment

belinski-roman, 2025-09-28:
> > You can create two labels per segment
> 
> [@belinski-roman](https://github.com/belinski-roman) what does this mean?

[image]

paullee0, 2025-09-28:
> 
> One thing that would be really useful, though, is a tool to invert the direction of lines, arcs, and circles without needing to delete and recreate them, while also preserving their constraints. Considering this, the line orientation is more like a constraint that you also invert.
> 
> I’ve noticed when working with Lattice2 that some features rely on orientation. Being able to flip directions like this would make the results more consistent across different cases.

luzpaz, 2025-12-03:
@PaddleStroke Would you mind weighing-in here ?

kaiwas, 2025-12-03:
> Someone also mentioned to me

Maybe it was me. Right here )))
I think it was somewhere else.
(I had to find this example)

https://github.com/FreeCAD/FreeCAD/issues/24342#issuecomment-3353473226

kaiwas, 2025-12-03:
I'd also like to say that the algorithms for determining wall direction currently work well. There's no need to change anything. Don't even try!

https://github.com/user-attachments/assets/430706b6-1f87-40e1-b8cc-cb1b4b2b6d44

The video demonstrates how it works in ArchiCAD. Wall construction is similar in FreeCAD. BUT! In ArchiCAD, I can quickly change the alignment of a wall fragment.
In FreeCAD, I can only do this at the sketch level (by changing the construction direction of the sketch fragment).
This is the key difference.

In ArchCAD, control is at the wall level.
In FreeCAD, control is at the sketch level.

leoheck, 2025-12-04:
yeah, I cannot disagree too, I am doing this repetitive dance every day.

paullee0, 2025-12-05:
> offtop [@PaddleStroke](https://github.com/PaddleStroke) I just want to show that offsetting nested objects can also be useful. I always do this when I'm marking out the "preparation" for the foundation base.

There seems to be a few related issues here - 

1.  This offset feature is indeed one thing that is handy, better still in my 'wishlist' is selecting 4 walls / room and offsetting for finishes layer, moving the wall layout and room dimension and the finishes just follow.  This is somehow achieved in ArchSketch but not exposed yet.
2.  With same algorithm, the room area (and other information) is automatically calculated.

3.  Still, the Direction of the Sketch edges is better shown for easier understanding of user
4.  I re-read a few time thread but not sure if currently it is the Sketch Offset tool which is unpredictable, right?

Thanks.

PaddleStroke, 2025-12-05:
> Yes. Yes. I remember. The program is simply designed to repeat the same operations twice. Click, select, do, Click, select, do, Click, select, do.
> This has already been implemented in hundreds of places, and there's no reason to do it any other way.
> 
> In this world, time is precious. And only FreeCAD users have so much of it that they can afford to waste it on repetitive, monotonous actions. And if something doesn't work, they try to find answers and write macros to speed up the simplest actions.
> 
> I wish the developers would test their work on at least some real-world problem. Maybe then they'd realize how slow everything is.

First you have a computer, nothing prevent you from setting up the environment to build freecad and make pull request. Any CAD user have enough brain to learn to develop. You just don't want to invest the time.

Second, what you ask for is ridiculous. It's 10 times easier to use twice the offset tool in the edge case where you have nested loops and want them to offset in some precise way rather than consciously drawing the loops CW or CCW.

You suggest that we implement a visual to see line orientation. So that then the user can draw his loops in the correct CCW/CW orientation so that it then offset as he wants? What the likelyhood that he will get it right on the first try? "Oh no I drew my first loop CCW when it should have been CW now it's not offsetting as I wanted! Well let's start over the drawing!"

Sure it sounds so much easier. Any other CAD software that does this?

FlachyJoe, 2025-12-05:
The main problem is tools accept wire with no homogeneous edge orientation.
They should fix the wire before the use:
```python
doc = App.ActiveDocument
sk = doc.Sketch
edges = sk.Shape.Edges

# find the ends of the wire
points = [] 
for edge in edges:
	# points are countable but vertexes are not
	points.extend([v.Point for v in edge.Vertexes])
ends = [p for p in points if points.count(p)==1]

closed = points
if ends :
	# close the wire
	closed = [ends[0], *[p for p in points if not p in ends], ends[1], ends[0]]

wire = Part.makePolygon(closed)

# create the face needed by the fixer
face = Part.makeFace(wire)

fixer = Part.ShapeFix.Wire(wire, face, 1e-6)
fixer.perform()
fixedWire = fixer.wire()

if ends:
	# retrieve the supernumerary edge
	edgesOpen = []
	for e in fixedWire.Edges:
		if (not (e.firstVertex().Point == ends[0] and e.lastVertex().Point == ends[1])
		and not (e.firstVertex().Point == ends[1] and e.lastVertex().Point == ends[0])):
			edgesOpen.extend([v.Point for v in e.Vertexes])
	fixedWire = Part.makePolygon(edgesOpen)

Part.show(fixedWire)
```

"Left" wall with a mixed-CW-CCW sketch (red) and the fixed wire (transparent)

[image]

FlachyJoe, 2025-12-05:
I never encounter a reverse orientation between inner and outer sketch elements. Do you have a file to share?

paullee0, 2025-12-06:
> > I never encounter a reverse orientation between inner and outer sketch elements. Do you have a file to share?
> 
> It's not difficult. Just draw two rectangles. One clockwise and one counterclockwise. Then use the "wall" tool in the BIM workbench. The clock doesn't matter, but the direction of the lines is important.

ArchWall use directly the Sketch 'internal geometries' so user has direct control over each edge.

There could be a few ways to show the possible issues to my memory:-
1. Check sketch.Shape.Edges, the orientation edge by edge, vertex by vertex
2. Try Draft Wire, randomly in area of -X / -Y

As the 'output' shape was found to be in some case 'unpredictable', ArchWall codes was modified that it no longer used a Sketch.Shape.Edges, but directly use Sketch's 'internal geometries' which user directly input oneself.

paullee0, 2025-12-06:
[Sketch_Test_paul.FCStd.zip](https://github.com/user-attachments/files/23972950/Sketch_Test_paul.FCStd.zip)

```
s=Gui.Selection.getSelection()[0]
for e in se:
   Part.show(e)
   print(' edge..Vertexes[0].orientation - ', e.Vertexes[0].Orientation)
   ev = e.Vertexes[0]
   Part.show(ev)
```

This test show the Sketch.Shape.edges sequence is not following the Sketch internal geometries sequence (ID) - 
1. Sketch.Shape.Edges  edge 4, edge 5, edge 6, edge 7, edge 8 
are
2. Sketch internal geometries  ID 8, ID 7, ID 6, ID 5, ID 4

No problem on the first 3 geometries and orientation in this test.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

FlachyJoe, 2025-12-06:
@kaiwas you confirm the CW/CCW quality of sketch inner edges are reported to the feature constructed on its shape. I agree with this. What I ask is a file where sketch inner edges are CW and shown CCW by an hypothetical tool.

@paullee0 edge order in a wire doesn't reflect the orientation. You can create a wire with
```python
import Part
V_ = App.Vector
e0 = Part.makeLine(V_(0,0,0),V_(0,1,0))
e1 = Part.makeLine(V_(0,1,0),V_(1,1,0))
e2 = Part.makeLine(V_(1,0,0),V_(1,1,0)) #revert edge
w = Part.Wire()
w.add(e0)
w.add(e1)
w.add(e2)
```

kaiwas, 2025-12-06:
> [@kaiwas](https://github.com/kaiwas) you confirm the CW/CCW quality of sketch inner edges are reported to the feature constructed on its shape. I agree with this. What I ask is a file where sketch inner edges are CW and shown CCW by an hypothetical tool.

I don't quite understand what I need to do and what hypothetical instrument you're talking about.
Please clarify (translation difficulties may occur).

------------------

Actually, besides the sketcher, the start and end values ​​play a role in some other tools.
For example, the array along the path in DRAFT.

[image]

You can specify an offset from the start or end.

But how can I quickly understand where the line starts and ends without using the array values?

If everyone isn't tired of it yet, I can try collecting all the cases where the start and end values ​​matter. It's precisely because of these cases that visualizing the direction and the ability to reverse will be important.

kaiwas, 2025-12-07:
So, the situation is this.
For many BIM tools, only the direction of the first line is important.
But for walls, the direction of EVERY line is important.

What happens if we change the direction of the lines marked with number 1 in the figure?
Let's check.

https://github.com/user-attachments/assets/c5cb3e98-f863-47d9-9542-c9c547209cd3

As we can see, the geometry flipped.
And if I had the tool from this issue: https://github.com/FreeCAD/FreeCAD/issues/11957
I would have changed the direction very quickly.

All my examples are primarily related to BIM. But I've also seen examples in other workbenches where the direction of the lines mattered.

kaiwas, 2025-12-07:
Another example.
Attachment.
Incidentally, one of the most frequently used tools.
Depending on the direction of the lines, it attaches to different places.

Yes, I know you can activate a second landmark and rotate the sketches (that would be a solution). But all this is very complicated, and it would be easier to simply change the direction of the lines.

https://github.com/user-attachments/assets/e2241fb2-71cb-4daa-858e-48102a110b51

FlachyJoe, 2025-12-07:
ArchWall could works on SubElements instead of a full sketch then you could create as many wall as needed alignment for a unique sketch.

paullee0, 2025-12-07:
> UPD2/ Perhaps a new "tool" specifically for walls will be needed that will change all wall parameters for a selected segment (without switching to the sketch editing level). There are some developments in the ArchSketch addon ([@paullee0](https://github.com/paullee0) ). But I don't like the implementation: each alignment parameter is a loop. And the parameters aren't visualized (there's no help in the taskbar). Besides, such important tools should be available out of the box, without any add-ons.

Yes, I planned to do a list box so user can directly selected which Align is desirable (Left, Right, Center).

> For example, working with wall alignment in ArchiCAD. ([@paullee0](https://github.com/paullee0) ) Alignment changes occur instantly with a single button. In this case, the direction of wall construction doesn't particularly matter because changes can be made very quickly (without going to the sketch editing level or a complex list).

OK, I see how ArchiCAD works with alignment, thanks.