FreeCAD/FreeCAD #24254: SubtractivePipe: "Tool shape is not valid for boolean operation" error in 1.0, incorrect shape in 1.1.0dev

marbocub, 2025-09-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

- FreeCAD 0.21.2 → works correctly (original design environment)
- FreeCAD 1.0.2 → recompute fails with error
- FreeCAD 1.1.0dev rev14555 (2025-09-24) → no compute error, but wrong shape calculated

[SubtractivePipe-issue.zip](https://github.com/user-attachments/files/22559368/SubtractivePipe-issue.zip)

1. The following model was designed with FreeCAD 0.21.2 using SubtractivePipe.
[image]

2. When opening the same file in FreeCAD 1.0.2 and recomputing, the following error occurs:
Tool shape is not valid for boolean operation
[image]

3. When opening the same file in FreeCAD 1.1.0dev rev14555 (2025-09-24), the error no longer occurs, but the calculated shape is incorrect.
[image]

Result in 1.1.0dev rev14555:
[image]

I found the following issue that shows a similar error in 1.0. It seems related to adaptive functions in 1.0, but in my case 1.1.0dev introduced another issue, where an incorrect shape is calculated:
https://github.com/FreeCAD/FreeCAD/issues/18236

I also found the following PR, which has already been merged. It would prevent the error that occurred in 1.0, but in 1.1.0dev rev14555 it exposes a new issue, where an incorrect shape is calculated:
https://github.com/FreeCAD/FreeCAD/pull/23102

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Japanese/Japan (ja_JP)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Assembly4 0.50.18
  * CfdOF 1.30.2
  * CfdOF_fork 1.30.4
  * fasteners 0.5.38

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Japanese/Japan (ja_JP)
Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Trackball/Object center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/93.1131/1.5
Installed mods: 
  * Assembly4 0.50.18
  * CfdOF 1.30.2
  * CfdOF_fork 1.30.4
  * fasteners 0.5.38
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

marbocub, 2025-09-28:
I found that the issue where the orientation of a DatumPlane differs between 1.0 and 1.1.0dev when opening old data may have been introduced by #18332. Rev39622 (`c5fbbb3` in #18332; Dec 14, 2024) and earlier show the same orientation, while rev39623 (`4066100` in #18332; same date) and later show a different orientation.

#18332 seems to have changed the default orientation of DatumPlane, and the compatibility layer might not fully handle the conversion from old-style data to the current format used in 1.1.0dev.

maxwxyz, 2025-09-28:
@PaddleStroke could this be an attachment issue with the changes axis orientation early in 1.1?

PaddleStroke, 2025-10-01:
> @PaddleStroke could this be an attachment issue with the changes axis orientation early in 1.1?

Axis orientation changes early in 1.1 were reverted by Werner because he found a way to make it work while keeping the old inconsistent axis placements, and because it was causing backward compatibility issues. So those changes are gone.

The rotation change could be due to how attacher is resolving placements. I need to check this to understand what's going on

PaddleStroke, 2025-10-01:
@marbocub I see you tested right before https://github.com/FreeCAD/FreeCAD/pull/18332. But did you test also right after it? Because between 2024-12-14 and 2025-09-24 there's a lot of time

marbocub, 2025-10-01:
Yes, I tested the following commits, along with many others, to try to find the starting point of the issue:

* aed305d: before #18332; correct shape; 2024-12-14
* c5fbbb3: in #18332; correct shape
* 4066100: in #18332; incorrect shape
* 7620e0a: last of #18332; incorrect shape; 2024-12-14
* 7cfa0df: last of #18704; incorrect shape; 2025-01-02
* a67080a: last of #20026; incorrect shape; 2025-03-06

I’ve uploaded a full example (36MB):
https://raw.githubusercontent.com/marbocub/full-example/refs/heads/freecad-issue/full-example.FCStd
This is a duct of a 3D printer.
The previous example was just a part of this full example.

Sketch072, based on DatumPlane024, represents the cross-section of the duct.

[image]

[image]

[image]

[image]

---

When I restored the legacy code below, DatumPlane024 and Sketch072 ended up oriented differently. As a beginner, I found it challenging to handle this properly.
```
TopoShape AttachEngine::handleLegacyDatum(App::DocumentObject* obj)
{
    TopoShape shape;

    if (obj->getTypeId().isDerivedFrom(Base::Type::fromName("PartDesign::Plane"))) {
        Base::Console().warning("%s: PartDesign::Plane\n", obj->getFullName());
        auto geof = extractGeoFeature(obj);

        // obtain Z axis and origin of placement
        Base::Vector3d norm;
        geof->Placement.getValue().getRotation().multVec(Base::Vector3d(0.0, 0.0, 1.0), norm);
        Base::Vector3d org;
        geof->Placement.getValue().multVec(Base::Vector3d(), org);
        // make shape - an local-XY plane infinite face
        gp_Pln plane = gp_Pln(gp_Pnt(org.x, org.y, org.z), gp_Dir(norm.x, norm.y, norm.z));
        TopoDS_Shape myShape = BRepBuilderAPI_MakeFace(plane).Shape();
        myShape.Infinite(true);

        shape.setShape(myShape);
    } else if (obj->getTypeId().isDerivedFrom(Base::Type::fromName("PartDesign::Line"))) {
        Base::Console().warning("%s: PartDesign::Line\n", obj->getFullName());
        auto geof = extractGeoFeature(obj);

        // obtain X axis and origin of placement
        // note an inconsistency: App::Line is along local X, PartDesign::DatumLine is along
        // local Z.
        Base::Vector3d dir;
        geof->Placement.getValue().getRotation().multVec(Base::Vector3d(1.0, 0.0, 0.0), dir);
        Base::Vector3d org;
        geof->Placement.getValue().multVec(Base::Vector3d(), org);
        // make shape - an infinite line along local X axis
        gp_Lin line = gp_Lin(gp_Pnt(org.x, org.y, org.z), gp_Dir(dir.x, dir.y, dir.z));
        TopoDS_Shape myShape = BRepBuilderAPI_MakeEdge(line).Shape();
        myShape.Infinite(true);

        shape.setShape(myShape);
    }

    return shape;
}
```
And insert it in the AttachEngine::readLinks():
```
       auto shape = handleLegacyDatum(objs[i]);
        if (shape.isNull()) {
            Base::Console().warning("%s: is not Legacy Datum\n", objs[i]->getFullName());
            auto shape = extractSubShape(objs[i], subs[i]);
        }
```
The result shows DatumPlane024 and Sketch072 oriented differently:
[image]

PaddleStroke, 2025-10-02:
Ok thanks for pinpointing it! I'll check this now

PaddleStroke, 2025-10-02:
I think I nailed down the issue: 

Before attacher had a custom logic to find placement of App::Plane : 
```
        if (geof->isDerivedFrom(App::Plane::getClassTypeId())) {
            // obtain Z axis and origin of placement
            Base::Vector3d norm;
            plc.getRotation().multVec(Base::Vector3d(0.0, 0.0, 1.0), norm);
            Base::Vector3d org;
            plc.multVec(Base::Vector3d(), org);
            // make shape - an local-XY plane infinite face
            gp_Pln plane = gp_Pln(gp_Pnt(org.x, org.y, org.z), gp_Dir(norm.x, norm.y, norm.z));
            myShape = BRepBuilderAPI_MakeFace(plane).Shape();
            myShape.Infinite(true);
        }
```

After https://github.com/FreeCAD/FreeCAD/commit/4066100a79c3e6e82682f004d69133c794cb87aa we used the custom logic that is in `PartFeature::getTopoShape`:

```
else if (linked->isDerivedFrom<App::Plane>()) {
    static TopoDS_Shape _shape;
    if (_shape.IsNull()) {
        auto plane = static_cast<App::Plane*>(linked);
        Base::Vector3d dir = plane->getBaseDirection();
        BRepBuilderAPI_MakeFace builder(gp_Pln(gp_Pnt(0, 0, 0), Base::convertTo<gp_Dir>(dir)));
        _shape = builder.Shape();
        _shape.Infinite(Standard_True);
    }
    shape = TopoShape(tag, hasher, _shape);
}
```

This was done to avoid code duplication.
Now while they look similar, the one in  `PartFeature::getTopoShape` is actually different.

It calls plane->getBaseDirection(). This function returns the normal of the plane, which corresponds to its local Z-axis. However, it completely ignores the rest of the plane's Placement rotation information, which defines how its local X and Y axes are oriented.

The code then creates an OpenCASCADE plane using the constructor gp_Pln(Origin, Normal). When you provide only a normal vector, OpenCASCADE is forced to create a valid, but **arbitrary**, coordinate system for the plane. It will pick some direction for the plane's internal X-axis that is perpendicular to the normal. This choice does not respect the original App::Plane object's full orientation.

Then what happens in your file is that you have a attachmentoffset rotation. You say 'rotate by 26deg along Y axis'. But now the X and Y are not defined properly in getTopoShape code. And so it rotates along the X axis instead of the Y.

PaddleStroke, 2025-10-02:
@marbocub meanwhile you can just change the attachment offset of -26 from the Y axis to +26 X axis and that should fix your plane.

PaddleStroke, 2025-10-02:
It appears I got confused and my previous analysis not accurate.

So here is what is actually happening : 

- The new code `PartFeature::getTopoShape` : 
It creates a static shape for all App::Plane. That is actually correct. `dir = plane->getBaseDirection()` is actually just a Z axis (0,0,1). So the static shape is correctly defined as a standard XY plane.
THEN getTopoShape transform this shape to the correct placement.

So this behavior is the correct behavior. **The placement is applied to a standard XY plane**

- Before, the old code in attacher :
It was creating a plane from a transformed direction. ie it was finding the transformed direction `plc.getRotation().multVec(Base::Vector3d(0.0, 0.0, 1.0), norm);`. **The placement was applied ONLY to the direction. Resulting in a plane shape that had the same Z direction, but not the same X direction as the original.**

So this behavior was bad.

So the conclusion is that the current code is what we want. The old code was wrong.

Who is affected?
- Files created before https://github.com/FreeCAD/FreeCAD/pull/18332
- With objects attached to a App::Plane (YZ or XZ)
- That have a attachment offset with some X/Y rotation.

Now what to do about files that have the same issue? I don't know, the old code was wrong so I am unsure what we should do. Restablishing the wrong code seems like a bad idea. And I see no way we can detect to migrate.

marbocub, 2025-10-03:
This may or may not be directly related, but I noticed something about creating a Part::DatumPlane in v1.1.0dev.

When I create a datum plane by:

1. Setting the position with a vertex  
2. Setting the orientation with a reference plane  

…the orientation in Part::DatumPlane of v1.1.0dev is easier to understand than in PartDesign::Plane of v1.0 or v0.21, as the axes now align, but the Z axis is flipped, which makes sketching slightly less convenient.

I have attached a zip file containing two example files for reference:  
 [Plane_comparison_v1.0_vs_v1.1dev.zip](https://github.com/user-attachments/files/22688761/Plane_comparison_v1.0_vs_v1.1dev.zip)
- PartDesign_Plane_v1.0.FCStd  
- Part_DatumPlane_v1.1dev.FCStd  

Screenshots:  
* PartDesign_Plane_v1.0.FCStd: +Y is up  
[image]

* Part_DatumPlane_v1.1dev.FCStd: –Z is up  
[image]

From my experience as a user, it would be easier if:  
- XY_Plane → +Y is up  
- XZ_Plane → +Z is up  
- YZ_Plane → +Z is up

---

I also want to share a personal hope regarding compatibility.  
I think one of FreeCAD’s strengths is long-term file compatibility.  
As a user, I would be very happy if both a compatibility layer for PartDesign::Plane from v1.0 and earlier could be maintained, and Part::DatumPlane could behave in an ideal way in v1.1 and later versions.

marbocub, 2025-10-15:
I expanded the verification scope and found that this issue has a much broader impact, affecting most objects attached in mmTangentPlane mode, including Part objects and Sketches.

I’ve attached two example files: [example-sketch-and-part-object.zip](https://github.com/user-attachments/files/22919647/example-sketch-and-part-object.zip)

example-sketch.FCStd:
[image]
[image]

example-part-object.FCStd:
[image]
[image]

pieterhijma, 2025-10-24:
> Now what to do about files that have the same issue? I don't know, the old code was wrong so I am unsure what we should do. Restablishing the wrong code seems like a bad idea. Also if we revert to that bad code, files created since 18332 would then become corrupted... And I see no way we can detect to migrate. [@chennes](https://github.com/chennes) [@hyarion](https://github.com/hyarion) [@kadet1090](https://github.com/kadet1090) [@FreeCAD/code-quality-working-group](https://github.com/orgs/FreeCAD/teams/code-quality-working-group)
> 
> Any idea?

I find it really hard to follow what exactly happened. I assume it is about [this code that has been removed](https://github.com/FreeCAD/FreeCAD/pull/18332/commits/4066100a79c3e6e82682f004d69133c794cb87aa#diff-39390d1d67bdf8c90422df5552474c42d127d8747cda9343a457c1a73bb30d4aL847-L858). My understanding is now that that was a special case to create an App::Plane. This has now been generalized and we take the original toposhape that now supports fully qualified subnames (before it did not?). I'm trying to understand why the original code was there in the first place. 

In any case, I'm of the opinion that we should not restore old incorrect code but always move forward with the most correct code.

pieterhijma, 2025-10-29:
> [@pieterhijma](https://github.com/pieterhijma) Thank you for following up on this issue.
> 
> I've submitted a pull request ([#24529](https://github.com/FreeCAD/FreeCAD/pull/24529)) that introduces a compatibility layer to convert legacy data created in pre-1.1.0dev versions to match the current behavior. Rather than restoring the old code paths, the approach modifies legacy data during document loading so that it conforms to the updated logic in 1.1.0dev.
> 
> Please note that the current approach in [#24529](https://github.com/FreeCAD/FreeCAD/pull/24529) differs from the initial proposal. While this approach may not be perfect, I believe it can be practically helpful in many cases.
> 
> The PR is currently awaiting review. I hope this helps move the discussion forward.

@marbocub, thanks for the explanation. I was aware of the PR and I'm one of the reviewers.

However, before I review that, I'm first interested in understanding the whole context. For instance, I would like to understand how to verify the following statements from @PaddleStroke:

> So here is what is actually happening :
>
> * The new code `PartFeature::getTopoShape` :
>       It creates a static shape for all App::Plane. That is actually correct. `dir = plane->getBaseDirection()` [...]

@PaddleStroke, could you perhaps explain the history? I believe there was also code from Werner. I'm just wondering why there has been old code that was incorrect for so long.

PaddleStroke, 2025-10-30:
@pieterhijma you will need to read the code before and after to verify my statements you quoted.
Basically attacher was implementing it's own custom 'get topo shape' code for App::Planes. And this custom way of doing that was wrong because it was only transforming the Z axis, not the whole placement. So then the X axis ended up relying on the gp_plane ctor.