FreeCAD/FreeCAD #24261: PartDesign: Dragger goes nuts again (False negative)

leoheck, 2025-09-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

A simple sketch (circle) that is attached to a DatumPlane is producing a bad dragger. 
This is because I checked the skip recomputation in the given file.
However, if I accept the feature, it works. 

[image]

Steps to reproduce.

1. Mark file with Skip Recomputation
2. Add a feature to an existing body
3. Observe the issue
4. Hit Ok.
5. It produces a fine feature.

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
Installed mods: 
  * Assembly4.1 0.60.2-0.1
  * Cables 0.2.1
  * CurvesWB 0.6.64
  * FreeCAD-render 2024.12.15
  * Assembly4 0.60.2
  * sheetmetal 0.7.55
  * DynamicData 2.75.0
  * Curves 0.6.70
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * Assembly4.1_mod 0.60.2-0.1
  * OpenTheme 2025.5.20
  * fasteners 0.5.42
  * Silk 0.2.0
  * osh-autodoc-workbench 0.2.3
  * CurvedShapes 1.0.13
  * freecad-povray-render
  * OpticsWorkbench 1.2.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.7.6
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leoheck, 2025-09-26:
I created a Sketch and recomputed it.
Then, with the Sketch selected, I created a Pad.
The dragger appears in an unset position.
But the preview didn't show up.

With the Pad task still open, I recomputed it in the Object Tree (right-click → Recompute).
Then the preview was updated fine, but the dragger still looks wrong.

So there are 2 issues in one when we use Skip Recomputes.

1. Dragger does not work properly.
2. The preview does not appear.

[image]

captain0xff, 2025-10-16:
The reason you are getting correct behavior when the task panel is changing is because the feature is providing the correct info.

leoheck, 2025-10-16:
> The reason you are getting correct behavior when the task panel is changing is because the feature is providing the correct info.

The skip recompute could keep the preview working then.

captain0xff, 2025-10-17:
Which means the geometry is being recomputed.