FreeCAD/FreeCAD #24264: Sketcher: Offset does not work fine with Arc Slot

leoheck, 2025-09-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I am not sure I am doing something that cannot be geometrically possible. 
I want to offset a new geometry inside an arc slot.

But it looks like the Offset does not work 100% of the time with Arc Slots (with maybe external geometry).

Sample file: 

[offset_issue_arc_slot.FCStd.zip](https://github.com/user-attachments/files/22563629/offset_issue_arc_slot.FCStd.zip)

This is the issue:

https://github.com/user-attachments/assets/e048d09e-7ca9-4662-a5db-06ddab1a656b

This is what I wanted to achieve; it does not look impossible.

https://github.com/user-attachments/assets/002ea080-7eed-47cd-9391-ff0cbd1a344c
 

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
Installed mods: 
  * Assembly4.1 0.60.2-0.1
  * Cables 0.2.1
  * CurvesWB 0.6.64
  * FreeCAD-render 2024.12.15
  * Assembly4 0.60.2
  * sheetmetal 0.7.55
  * DynamicData 2.75.0
  * Curves 0.6.70
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * Assembly4.1_mod 0.60.2-0.1
  * OpenTheme 2025.5.20
  * fasteners 0.5.42
  * Silk 0.2.0
  * osh-autodoc-workbench 0.2.3
  * CurvedShapes 1.0.13
  * freecad-povray-render
  * OpticsWorkbench 1.2.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.7.6
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

theo-vt, 2025-09-30:
Hi hello! I can replicate with the given file, but I don't think that it is an issue with the arc slot per say:

https://github.com/user-attachments/assets/485e1b95-7b7b-4820-9995-c2e7beef646c

maxwxyz, 2025-09-30:
@PaddleStroke FYI

PaddleStroke, 2025-09-30:
The Linear pattern and scale tools do not fails like the polar pattern. So I think the external arc that is buggy is reversed. Which probably somehow makes OCC fails to handle it in the offset function.

And the polar pattern is rotating the arc in the wrong direction because it's reversed.

PaddleStroke, 2025-09-30:
So the root problem is that the external geometry tool should make sure that the created geometries are correct. ie that the arcs are not reversed.

On top of that we can add a failsafe to offset and polar pattern to fix any bad arc input.

PaddleStroke, 2025-09-30:
This is similar as https://github.com/FreeCAD/FreeCAD/pull/24053, but here the bad arcs are created by the external geometry tool.

PaddleStroke, 2025-10-01:
So after a long serie of test, it appears that the offset failure has nothing to do with reversed arcs (which still needs to be fixed though). As I managed to create normal arcs where this situation occurs as well. Just moving a little bit the geometries and then offset works.

PaddleStroke, 2025-10-01:
Even better file : 

[offset_issue_arc_slot_3.zip](https://github.com/user-attachments/files/22635368/offset_issue_arc_slot_3.zip)

Two identical arc slots. Positioned at exactly the same position (as I created it by snapping the second arc slot into position). No reversed arcs. As far as I can tell they are identical.

Yet if you offset : 
- edge1-4 (first slot) -> bug
- edge5-8 (second slot) -> OK

Why? I don't know...