FreeCAD/FreeCAD #24274: PartDesign: symmetric pad has extra seamline breaking existing models

jffmichi, 2025-09-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Using PartDesign Pad in symmetric mode without refine enabled causes an extra seamline in the current main compared to e.g. 1.0.2. This can cause existing models to break, e.g. the chamfer in the provided example file. To reproduce just do a full recompute of the example file.

[image]

[partdesign_pad_seam.zip](https://github.com/user-attachments/files/22569510/partdesign_pad_seam.zip)

### Full version info

```shell
current dev version
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

depthoffocus, 2025-09-27:
I can certainly reproduce this, and it does the same in two-dimensions mode, though here I feel a bit less sure it's incorrect behaviour:

[image]

```
OS: macOS 15.7
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Free Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.75/2
Installed mods: 
  * ThreadProfile 1.97.0
  * freecad.gears 1.3.0
  * OpenTheme 2025.5.20
  * fasteners 0.5.42
  * lattice2 1.0.0
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.55
  * Curves 0.6.70
```

maxwxyz, 2025-09-27:
It seems refine is turned off but by user preference and default it should be enabled.

depthoffocus, 2025-09-27:
> It seems refine is turned off but by user preference and default it should be enabled.

It was only enabled by default with 1.0, though, I think. So there could be a fair number of pre-1.0 designs out there where this is a TNP problem?

I don't know if I understand the TNP mitigations well enough to understand how bad a problem this is. 

But for example a cylinder -- if you extrude it symmetrically before 1.1 with refine set to false, chamfer both of its faces, open it in 1.1 and then change the extrusion length,  there will be a TNP issue from the chamfers regardless of whether they were created together or separately.

A blocker?

https://github.com/user-attachments/assets/459120fe-7b08-4476-92af-8bc348aa6486