FreeCAD/FreeCAD #24276: A weird behavior of a constraint with a rounded rectangle

philip-marlowe, 2025-09-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Please find attached mp4 file to see what the problem is.

Steps to reproduce
1. start from Parametric Body
2. create a new Sketch
3. create a Rounded Rectangle 
4. add a Radius Constraint to a corner
Whatever radius you set in the constraint, it won't affect the sketch at all.
5. add a Radius Reference to any of the other corners
6. change the radius of the constraint created in the step 4., it starts working

This problem is not reproducible on 1.0.2 stable version.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Japanese/Japan (ja_JP)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/93.2607/1.5
Installed mods: 
  * Curves 0.6.64
  * sheetmetal 0.7.24
```

### Subproject(s) affected?

Sketcher

### Anything else?

https://github.com/user-attachments/assets/dfffb568-8bd5-4d29-8173-6f326e98dd86

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

depthoffocus, 2025-09-27:
Here is what I mean about enabling the grid so that you can see what is going on.

I don't think this is a _bug_, and I think it is perhaps only confusing in practice to users who have longstanding experience of the pre-1.1 behaviour.

Basically the intent of the change was to make sure that sketches more complex than this don't get unexpectedly reshaped if adding the first constraint would do so. 

I definitely think that it can be confusing -- it can be even more confusing if there's some other geometry on screen because the change has the effect of causing that geometry to appear to grow or shrink! 

But without the grid enabled I am not sure what presentational change would draw the user's attention to this to resolve the confusion that it can cause. 

Perhaps if the view scale changes significantly is significantly scaled when the first constraint is added, the scale indicator in the bottom right could be briefly highlighted with a colour change?

https://github.com/user-attachments/assets/83b523b8-a58a-4b1c-a719-1ed334b09894

philip-marlowe, 2025-09-27:
Thanks @depthoffocus ! Your explanation perfectly makes sense. Yes, some visual reactions could help to prevent confusions.

Two auto-scaling were introduced at the same time. One is for the geometries, the other is for the view. And they are only triggered by the first constraint on the sketch. Even a reference constraint is counted.

I got it.

Here is my rough idea.
When the auto-scaling happens, first it affects the geometries immediately, and then, the view zooms in/out with animation like View->Standard View->Fit All.

Sudden change of the view could be always confusing unless it was intentionally kicked by the user.