FreeCAD/FreeCAD #24281: Access violation occurs when adding the first constraint

philip-marlowe, 2025-09-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

From Report View:
```
15:25:24  Exception performing autoscale: Access violation
15:25:24  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Invalid constraint index: -1
15:25:24  Unnamed#Sketch: App.getDocument('Unnamed').getObject('Sketch').setDatum(-1,App.Units.Quantity('102.170000 mm'))
15:31:52  QTextBrowser: No document for file:///C:%5CUsers%5CXXX%5CDownloads%5CFreeCAD_weekly-2025.09.24-Windows-x86_64-py311%5Cdoc%5C/ThirdPartyLibraries.html
```

Please find attached mp4 file to see what's the problem.

Steps to reproduce:
1. start from Parametric Body
2. create a Sketch
3. create a Rectangular without constraints
4. add Distance Dimension to an edge
The constraint is not added, and errors are logged in the Report View.

Note:
The problem doesn't occur when a Distance Dimension is specified while creating the rectangle in step 3.

This is not reproducible on version 1.0.2 stable.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Japanese/Japan (ja_JP)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/99.4147/1.5
Installed mods: 
  * Curves 0.6.64
  * sheetmetal 0.7.24
```

### Subproject(s) affected?

Sketcher

### Anything else?

https://github.com/user-attachments/assets/0d62094d-9f41-41bf-be17-ab1ec36f00f8

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-09-27:
I can confirm.
@theo-vt FYI regarding auto-scale

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/133.772/2
Installed mods: 
  * AddonManager 2025.8.16
  * Cables 0.2.1
  * CurvedShapes 1.0.13
  * Defeaturing 1.2.6
  * dodo 1.0.1
  * freecad.gears 1.3.0
  * Gridfinity 0.11.9
  * Manipulator 1.6.0
  * OpticsWorkbench 1.2.2
  * Rocket 4.1.1
  * sheetmetal 0.7.24
  * Silk 0.2.0
  * Telemetry 1.0.2

```

FEA-eng, 2025-10-01:
I can't reproduce it. Can you check in the newest weekly? Seems fixed there.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/30 23:55:55
Build type: Release
Branch: main
Hash: b08a0f87d68dd38cfcf39046209a7075adf11576
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```