FreeCAD/FreeCAD #24290: CAM:  MillFace operation is terrible at mill facing

sliptonic, 2025-09-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The MillFace operation is old and based on pocket.  Over time, it has accumulated more and more features to the point that it is basically the pocket operation.  The one thing it doesn't do well is mill facing.

A mill facing operation is intended to  clear the top of the stock in preparation for additional operations.  The operation should work on the stock object not the underlying features of the part.

I think most of the options in MillFace should be removed.  It should be refactored to produce a reasonable face milling toolpath
- always ignoring interior holes.
- always extending by at least the radius of the cutter all the way around the stock boundary.
- always starting more than the radius of the cutter outside the polygon (facing tools rarely like plunging)
- It shouldn't support REST machining. (If you need that, use pocket)

It should have a small number of options to set and configure the pattern.

### Full version info

```shell
any
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

sliptonic, 2025-09-27:
There are a handful of strategies to consider for clearing

- spiral: moves around the area circling inward.
- zigzag: moves back and forth alternating between climb and conventional milling
- unidirectional: Moves from one side to the other, lifts, returns to the other end.
- bidirectional: Cuts in both directions but rapids from one side to the other to maintain climb/convential

[image]

sliptonic, 2025-09-27:
One more possible strategy occurs to me.  Call it 'true spiral'  

This would use G2/G3 arc moves to circle into the center of the part.  This would be very inefficient if the part doesn't have a roughly 1:1 aspect ratio.   If it's long and narrow, a lot of time will be spent cutting air.

I think this strategy should be considered later.  There may be desire to start at the center and move outward but that would require a tool that can plunge.  This additional complexity can wait.

[image]

sliptonic, 2025-09-27:
And here's a mockup of a UI

Feedback from discord is that angle and direction preference can be combined into just angle.  I agree.

[image]

Connor9220, 2025-09-28:
I’m still not clear on the Axis preference. You have the angle option which would do the same?  Could that be simplified?

sliptonic, 2025-09-28:
Thinking more about how to simplify the interface for directionality. 
A single control for angle could do it.  Consider the simple case of a the directional strategy. (These considerations are basically the same for all the strategies, but directional is the easiest to think about.)

[image]

It's easy to see how a single control for angle would control the direction anywhere between 0 and 360 degrees. But it's not apparent why 0 degrees goes the way that it does.  As a user, I can't anticipate whether 90 degrees will go top to bottom or bottom to top.

The control could just as easily go from -180 to +180  but that doesn't really help.  The user is going to have to experiment to figure it out.   Which kinda sucks.

Another alternative is two controls:  Angle (0-180) and a checkbox to 'reverse' the cut.
In the simple case above, the user could just check the box and the primary direction would flip but still maintain climb/conventional.  If the user puts in 90 and it goes the wrong way, they don't have to change the input to either 270 or -90, they just click the 'reverse' tick.

My intuition is that the most common cases are either parallel to the X axis or the Y axis, in one direction or the other.  I might be wrong.  Maybe users prefer the cuts to be aligned with the long axis of the part or 90 offset from it and there's no reason to assume the part will be aligned relative to the machine axis this way.

Connor9220, 2025-09-28:
That could be addressed with a graphic to indicate which way is 0 degrees. Which to me would be to the right. 90 at top 180 to the left and 360 to the bottom.

sliptonic, 2025-09-28:
[image]

polidram, 2025-10-02:
> One more possible strategy occurs to me. Call it 'true spiral'
> 
> This would use G2/G3 arc moves to circle into the center of the part. This would be very inefficient if the part doesn't have a roughly 1:1 aspect ratio. If it's long and narrow, a lot of time will be spent cutting air.
> 
> I think this strategy should be considered later. There may be desire to start at the center and move outward but that would require a tool that can plunge. This additional complexity can wait.
> <img alt="Image" width="2000" height="1443" src="https://private-user-images.githubusercontent.com/538057/494801701-0af5e7f4-53c9-42d8-8cfd-4fda6655718f.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTk0MjUxMjUsIm5iZiI6MTc1OTQyNDgyNSwicGF0aCI6Ii81MzgwNTcvNDk0ODAxNzAxLTBhZjVlN2Y0LTUzYzktNDJkOC04Y2ZkLTRmZGE2NjU1NzE4Zi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUxMDAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MTAwMlQxNzA3MDVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hY2Y5YTEzZDE1NmZmNWUzNWMxNzgyMDFjZDcxZjc1MmEzZjFkNzFhNzUzMzlmM2JkNjdkNzk5ZmY1NjA1MzVlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.m4LBSowOCo9breR-QsP3SInQBplmkyMeK68MPC6VUtU">

I’ve been performing the spiral facing strategy in FreeCAD in the following improvised way:

I create a spiral in the Part workbench, then add it to the Job. After that, I apply the Engraving operation (modified by Tarman to allow starting from outside the stock) on the spiral path.

It would be much better if we had a native spiral option for cylindrical parts in the MillFace operation.

[image]

In the video below, I show the operation in practice:
https://www.youtube.com/shorts/sY1BYnz62_4

opensourcecnc, 2025-10-05:
Being able to avoid interior holes is useful, in my opinion. Here's a use case: I want to flatten the stock to get flat edges. I don't want to waste time flattening all the central area, since I'm going to pocket it later. I know that I could do that using the pocket operation, but this brings us in the same situation as for the millface: using an operation for something that it wasn't intended to do.
I think a checkbox to avoid holes should be there.

I got here by trying to find out why in 1.0 or later it doesn't avoid the holes when using face region, because I didn't want to use a pocket instead

[image]

sliptonic, 2025-10-05:
Forget about the names we're using "Facing",  "Pocketing",  etc.  We need to be clear whether there's just one thing with different flavors or two distinctly different things.  

I believe that there are two things.   One of those things is model-aware.  The tool path depends on the specific geometry of the model and the configuration options that the user has set.  We currently call that thing "pocket" even if the area being cleared doesn't have walls.

The other thing is not model aware.  It works on the stock object.  It assumes that there are no walls and the tool cannot plunge so the tool path is always started from outside the area to be cleared and enters laterally. 

Part of the reason that millface became less useful is we tried to add model awareness to it.  This issue and the associated PR is going the other way.  The intention is to make mill facing really good at doing that one thing.  

If you need model-awareness to clear an area, then the pocket operation can do what you need.   Now, if you wanted to propose that the pocket operation needs alternative strategies or improvement to make clearing areas more efficient, I'll be the first to agree with you. 

I just don't think we should try to make mill facing handle all those use-cases.

opensourcecnc, 2025-10-05:
If you ask me, from a user perspective, I would prefer a totally new operation, such as Flatten or similar and keep the old millface as deprecated. This will avoid a lot of confusion

opensourcecnc, 2025-10-06:
This means that people used to the old MillFace will have some trouble understanding what's going on. Also, backwards compatibility will be pretty difficult to handle. Adding the new operation (the entire behavior will be new) under a new name would reduce this kind of confusion.

tarman3, 2025-10-06:
@opensourcecnc 
Glad to know that you tried my build
About CAM3 you can write in [forum](https://forum.freecad.org/viewtopic.php?t=98993)

sliptonic, 2025-10-06:
> I was talking just about the naming. It shouldn't be called the same
> or almost the same, as it will lead to confusion. A completely new name
> would be better.

We did some survey of existing CAM software and the term 'facing' most accurately reflects what this operation is meant to do.  Using a new name like 'flatten' will avoid confusion within FreeCAD between our old and new mill facing but will make FreeCAD less consistent with industry standard terminology.

So I guess it's preferable to confuse existing users temporarily rather than confuse everyone forever.

Connor9220, 2025-10-06:
> First thing is related to the final depth. I believe being able to select lines and vertices would be useful. For now, selecting anything other than faces is locked. If I accidentally set the final depth to a face inside a pocket, I cannot go back to the top face of the stock unless I change it's view->display mode to flat lines, so I can select the face and set the final depth value.
> 

That's odd, it shouldn't be able to go inside a pocket..

> The second thing is about the pass extension: it seems odd to only do it on two sides. If it's about the initial plunge, can't it be done using a lead-in dressup? this will reduce the milling time a lot. If indeed it is intended to mill some extra material around the shape, why not create the stock with the corresponding extensions in the first place? Wouldn't that be the more "natural" way? Have the freecad stock replicate the stock in real life?

We are do extensions automatically on the entry and exit sides of the cutter,  This is very typical. 

Fusion 360 has these options
Pass Extension: Distance to extend the passes beyond the machining boundary. -- We have this.

Stock Offset: Specifies the distance the machining boundary extends beyond the stock boundary in both X & Y - We don't have this yet.

Stock to Leave (Axial Stock To leave) = Material Allowance in FreeCAD (I would like to rename to Stock to Leave to match other Operations, specifying Axial or Z)

I set my stock to my material size, make it easier for probing and touching off, I don't want to extend the stock size.  I would rather have the operation be able to support that.

opensourcecnc, 2025-10-06:
Clicking on a face and then on the button next to final depth will set the Z value of that face as the final depth. And it will look like in my screenshot. Now I have two options: set the final depth manually or use the view properties of the stock and then select the top face and press again on the button next to final depth.
As for the stock offset, it might be a good option, but why would I want to mill outside of the material that's on the table? Other then defining a stock that doesn't replicate the real life stock, I don't see any use cases. But I agree that there are a lot of times when I define the stock much smaller than the real piece and would use the stock offset option...

[image]

closed 2025-11-28 by chennes