FreeCAD/FreeCAD #24310: Building release packages: Alert: No files found in translations*.*

yan695432, 2025-09-29:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

File: "D:\FreeCAD\FreeCAD-1.0.2\installer\Release\translations\*.*" -> no files found.
Usage: File [/nonfatal] [/a] ([/r] [/x filespec [...]] filespec [...] |
   /oname=outfile one_file_only)
!include: error in script: "setup\install.nsh" on line 60。

Where do the translations files come from?
Can I comment out the relevant line in setup\install.nsh?

### Full version info

```shell
OS: Windows 10
Architecture: x86_64
Version: 1.0.2
libpack:LibPack-1.0.0-v3.0.0-Release
Build type: Release
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-10-01:
@yan695432 can you reproduce with 1.1.0-dev ?

FEA-eng, 2026-08-27:
@yan695432 Please confirm if it's still relevant.

luzpaz, 2026-09-16:
No response after multiple attempts. Closing

closed 2026-09-16 by luzpaz