FreeCAD/FreeCAD #24335: Sketcher: Insufficient analysis accuracy in the Sketch Validator.

kaiwas, 2025-09-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

A comment prompted me to file this issue. https://github.com/FreeCAD/FreeCAD/issues/16254#issuecomment-3086296337

And with the very first simple test, I discovered that the tolerance in the analysis is clearly broken.

How to reproduce:
1) Draw a rectangle in the sketcher.
2) Assign dimensional constraints (any).
2) Select one vertex that contains the dimensional constraint and press DEL (unmerge the vertices).
3) Close the sketch and run validation.

I don't know what units the tolerance is specified in, but any of the outlier values ​​is not suitable for merging vertices. I was able to merge vertices by setting a very large value. However, sometimes it even found more than one unmerged vertex.

Video demonstration. I won't attach a test file, as everything is clearly visible in the video. At the beginning of the video, I validate a large sketch without a size constraint.

https://github.com/user-attachments/assets/6dee287b-cdee-4fda-a43b-9c9aa3095495

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * addFC-main 1.6.2
  * Reinforcement
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * LyapinWorkbrench
  * FreeCAD_FastenersWB-master 0.5.42
  * FreeCAD_SketchArch-master
  * FrameForge 0.1.3
  * Road 2025.2.22
  * Curves 0.6.70
  * freecad.gears 1.3.0
  * CurvedShapesWorkbench-master 1.0.13
  * sheetmetal 0.7.53
  * OpenTheme 2025.5.20
  * FreeCAD_ThreadProfile-master 1.97.0
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

MikePryadko, 2025-09-30:
Hi! It looks like problem but (in my simple test by your description) does not produce a **real** problem 😏 such sketch still "closed" and usable for pad/pocket operations:

https://github.com/user-attachments/assets/bbc851c9-a27f-4a9f-aa1b-50a3efab29ac

P.S. My FreeCAD version: `1.0.2`

kaiwas, 2025-09-30:
Perhaps the problem won't just be with dimensions.
I've discovered that symmetry can also play a role (aiming at the origin is very difficult). Video.

https://github.com/user-attachments/assets/29cc0017-bc07-4941-85a0-1b85567b3bbe

kaiwas, 2025-09-30:
> Hi! It looks like problem but (in my simple test by your description) does not produce a **real** problem 😏 such sketch still "closed" and usable for pad/pocket operations:

Oh yeah! And then you change the rectangle's size and everything breaks.

https://github.com/user-attachments/assets/0b9b0ce3-28d6-463e-9f02-9eb4306fe13d

MikePryadko, 2025-09-30:
@FlachyJoe but what about https://github.com/FreeCAD/FreeCAD/issues/24335#issuecomment-3351576808 ?

kaiwas, 2025-09-30:
Okay! I'd be happy to forget about the Validator as a misunderstanding if there was a convenient tool for merging "selected" vertices within the sketch editing mode. In this case, the user would be able to decide for themselves which vertices to merge, rather than trusting an algorithm that operates incomprehensibly at the level of the "entire" sketch. I'm talking about this https://github.com/FreeCAD/FreeCAD/issues/16254

kaiwas, 2025-09-30:
I understand that constraints and vertices are a complex matter. A "free" vertex should be merged with the closest one with a constraint.

[image]

kaiwas, 2026-10-27:
still relevant

maxwxyz, 2026-11-07:
There seems to be a fix in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/2e9e973137dc1df8e512d2134618522acd1a375e

The validator no longer treats every constraint as a coincidence, so dimensioned vertices stay in the gap check.